Ysurac/openmptcprouter #2530: Struggling to get client VPN (UDP) to work through OpenMPTCP

DigitalLabs420, 2022-09-11:
## Expected Behavior
Howdy! 

Sorry to be a bother, but I was just wondering whether anyone had a guide on how to get OpenVPN UDP VPN to work through OpenMPTCP. 

I've attempted OMR-BYPASS, in the sections "Port Destination" and "Port source"; however, not matter what adjustments I make, VPNs won't work. 

I'm also using V2RAY, and as mentioned in issue #1373, 
"For info, VPN in OpenMPTCProuter are used for anything that is not TCP. Your OpenVPN config use UDP, so this use the OpenMPTCProuter VPN.
You can also try to set v2ray as proxy, this proxy can do UDP." 

However, still no luck. 
## Current Behavior

So, the DDNS name does get resolved into the IP; however, it appears that the UDP traffic isn't working and thus results in a TLS Handshake timeout. 

Any help or insight would be appreciated. 

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: Binary Lane 
  - OpenMPTCProuter platform: x86_64

Ysurac, 2022-09-11:
I don't really understand your problem, you need to bypass OpenVPN server IP or not needed ?
By default OpenVPN in UDP or TCP mode can work over OpenMPTCProuter.

DigitalLabs420, 2022-09-11:
Apologies @Ysurac, I may not have been clear. 

Currently I'm unable to connect to another OpenVPN server when connected through OpenMPTCP, the current behaviour is that it just timeouts. 

If  I attempt the connection via my regular Openwrt router, it works fine (currently running dual NAT).  As per your answer, a OpenVPN TAP should just work?

Ysurac, 2022-09-11:
Are you using it over TCP or UDP ?
I use OpenVPN UDP using TUN interface everyday for my daily job, I already tried TCP also without issues.
You can connect or not at all ? You VPS provider have a firewall or not ?

DigitalLabs420, 2022-09-14:
> 

I have a guess it's related to the broken state of NTP causing TLS issues. 

I'll let you know how I go

ahayes, 2022-09-24:
@DigitalLabs420 If I understand your problem correctly, I had a similar problem that I was able to resolve in my OpenVPN configuration.

I have an Ubuntu Linux system on my LAN running AirVPN Suite (uses bluetit OpenVPN3) that could not connect after I replaced my old Edgerouter with my new OpenMPTCProuter (5.15 version) box. I added `airproto tcp` to the bluetit.rc file and everything started working. The AirVPN Suite docs state:
`airproto: (string) Protocol to be used for automatic boot connection. Possible values: udp, tcp. Default: udp`

I don't know how that translates to your setup, but the docs for my VPN stack are here if you want to have a look: https://airvpn.org/suite/readme/

closed 2022-12-28 by github-actions[bot]