Ysurac/openmptcprouter #2532: |BUG] [VPS 0.1028]

Kalimeiro, 2022-09-11:
## Expected Behavior

When change VPN between OMR and VPS on Wizard, the file /etc/shorewall/params.vpn need to be updated correctly.

## Current Behavior
When change VPN, like Glorytun UDP to Glorytun TCP, the VPS_ADDR var contain the IP of Glorytun UDP (10.255.254.1) insteed Glorytun TCP (10.255.255.1) the VPS_IFACE var is define as "gt-udp-tun0" insteed "gt-tun0" and the OMR_ADDR var is the only one correct (10.255.255.2)

## Specifications

  - OpenMPTCProuter version: 0.59.1
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI4
  - Country: FR

[image]

[image]

Ysurac, 2022-09-11:
VPS_ADDR and VPS_IFACE are not used, so they are not updated.

Kalimeiro, 2022-09-13:
Ok you need to delete this var if unused, i close this issue.

closed 2022-09-13 by Kalimeiro