Ysurac/openmptcprouter #2535: [BUG] [0.59.1] When upgrading from 0.59 rc1, port forwarding not works.

Kalimeiro, 2022-09-11:
## Expected Behavior
When upgrading OMR 0.59 rc1 to 0.59.1, port forwarding works.

## Current Behavior
Currently, all port forward not working.

## Possible Solution
I don't have idea, the file /etc/shorewall/rules is correct and contain all my rules.

syslog don't log shorewall drop port.

## Specifications

  - OpenMPTCProuter version: 0.59.1
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI4
  - Country: FR

I need to make some test on VPS, but i think it's a bug related on ROUTER parts.

Kalimeiro, 2022-09-11:
i send you a mail with result of "uci shore firewall" and the rules file of shorewall.

Ysurac, 2022-09-11:
I need the result of this: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding#debug

Kalimeiro, 2022-09-13:
**Check that the VPN is running in System->OpenMPTCProuter and Status tab.**

[image]

**Check on the VPS in /etc/shorewall/rules that rules you set are available**

You already have the file by mail including "uci show firewall" from the router.

**Check on the VPS in /etc/shorewall/params.vpn that OMR_ADDR is the router side VPN IP (10.255.255.2 for glorytun TCP, the default VPN)**

[image]

**Check on the router if packets are coming on port 8080 from the VPN for example: tcpdump -i tun0 port 8080**

[image]

---

My rules doesn't need SNAT, and, working great before upgrade to final release.

Ysurac, 2022-09-13:
The 8080 was an example, would be better to test a port that you try to forward and check by connecting from outside VPS.
Then if you can see traffic on the router using tcpdump, check on your client destination.
If you see nothing on destination, I will need a "iptables-save | grep <port destination>" from the router.

Kalimeiro, 2022-09-13:
[image]

Kalimeiro, 2022-09-13:
> I will need a "iptables-save | grep " from the router.

grep ????

Ysurac, 2022-09-13:
iptables-save | grep 'destination port'
Did you try a tcpdump on client computer ?

Kalimeiro, 2022-09-13:
i have solved the problem with a compare file of /etc/config/firewall from a 0.59.1 port forward working and this 0.59.1, i check and tell here what is the problem.

Kalimeiro, 2022-09-13:
Ok, so the difference is i have this to the failed firewall configuration : 

```
config include 'gre_tunnel'
        option path '/etc/firewall.gre-tunnel'
        option reload '0'

config include 'v2ray'
        option path '/etc/firewall.v2ray-rules'
        option reload '0'

config zone 'zone_vpn'
        option name 'vpn'
        option masq '1'
        option input 'REJECT'
        option forward 'ACCEPT'
        option output 'ACCEPT'
        list network 'omrvpn'
        list network 'omr6in4'
        option mtu_fix '1'
        option auto_helper '1'
```

i have totaly removed "zone_vpn" and change reload "0" by "1" on gre_tunnel and v2ray, according to the 0.59.1 port forwarding working configuration, and instantaly, all port forward works.

closed 2022-09-15 by Kalimeiro

Ysurac, 2022-09-15:
You have another zone for VPN ? "zone_vpn" is required as name in several places.