Ysurac/openmptcprouter #2537: Failed update to 0.59.1

islander261, 2022-09-13:
## Expected Behavior

OMR should work after update to 0.59.1 and 0.1028 on VPS. Updated VPS first from terminal then reboot and then updated OMR from System > Sysupgrade on GUI. Both OMR and VPS rebooted multiple times in attempt to get things working.

## Current Behavior

OMR doesn't connect to VPS.
[image]

[image]
[image]

[image]

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: Can't tell OMR and VPS report different versions
  - OpenMPTCProuter VPS provider:Vultr
  - OpenMPTCProuter platform: RPI4

Ysurac, 2022-09-13:
VPS release is the one displayed on VPS, the router get the version from the API, it can't connect to API, so...
Check that server API key defined in the wizard is the same in ```/etc/openmptcprouter-vps-admin/omr-admin-config.json```, "openmptcprouter" section as "user_password".

islander261, 2022-09-13:
Ok, key reported on VPS in root@lopezcloud:~# openmptcprouter_config.txt (the one I have set in OMR) is not the same as the one in root@lopezcloud:/etc/openmptcprouter-vps-admin/omr-admin-config.json. Which one should I change? TIA

Ysurac, 2022-09-13:
The key in the json is the only one read by the API, the txt is only for information.
Both key should be same, but maybe an error somewhere...

islander261, 2022-09-13:
Thank you, all is working correctly now when using correct key.

Ysurac, 2022-09-13:
ok, now I need to be able to reproduce this issue to solve it...

closed 2022-12-18 by github-actions[bot]