Ysurac/openmptcprouter #2539: Will it work with the NanoPi R5S?

yitv, 2022-09-14:
Could you please confirm that OMR works with NanoPi R5S? Has anyone tested this? Would you mind guiding me on how to install it on nanoPI R5S model? 

I appreciate your help.

Ysurac, 2022-09-15:
I only support board that I own, I don't have it so it's not supported for now.
To add support, some patches will be needed.

yitv, 2022-09-20:
Thank you @Ysurac for the reply, sorry for the delayed response

closed 2022-12-25 by github-actions[bot]

kpma1985, 2023-08-26:
worked for me on r5s

lollover72, 2024-08-13:
i try to install openmtcp router on NanoPI R5S with these 2 metod but both fails:  i try to flash micro sd and boot from card, but doesn't work...blank screen, no signal and sys led and lan 2 led become on; i try also to install via usb flashing the eMMC but some result....can you post how your install procedure ? anyone success with  him?