Ysurac/openmptcprouter #2540: OpenMPTCProuter Status Network Overview "No Data"

Network-Traditions, 2022-09-15:
## Expected Behavior
When opening the "Status" tab of the System-OpenMPTCProuter menu of the web dashboard, the "Network overview" section will display a layout of the configuration with relative information about each device that comprises the system.  Internet sources, the OpenMPTCProuter device, the VPS server and the connected workstation.

Similarly, the "Dashboard" page will typically be populated with information about the version of the software, load of the device, uptime, proxy traffic, VPN traffic and total traffic followed by IP address and connection status for IPv4 & v6 in the upper left quadrant with DHCP devices identified in the lower left quadrant.  In the upper right quadrant, system information will be displayed identifying uptime, local time, load average, kernel version, model, architecture and firmware version.

## Current Behavior
When selecting this page, "No data" is displayed under Network overview.  Nothing else is displayed.
Reviewing the system log as a novice did not reveal

Similarly, data is missing from the "Dashboard" page as well.  Lower quadrants of this page contain no content within the gray squares, while the "Internet" (upper left quadrant) has all the Title labels ending with colons and a "-" where the data would be displayed.  "IPv4 Internet" shows "Connected: NO" and "IPv4: - " with the same for "IPv6 Internet".  The upper right "System" quadrant, however, appears to properly display the details as described above consistent with the working version of this setup as identified below.

## Possible Solution
Possible solution unknown, however the following tests were conducted to provide insight into the issue.
This system has been setup to mirror a [working system using v0.59beta6-5.4](https://github.com/Ysurac/openmptcprouter/issues/2365#issuecomment-1140081853) with the same ISP sources.  This system properly displays data for "Network overview" and Dashboard.  Comparing these two system show they are consistent with each other in regards to System: Software, Startup and Mount Points.  The Services and Network configurations are consistent as well.

While the aformention data is not displayed, the system does appear to function.  The interfaces for the T-Mobile and Starlink services successfully acquire a "Sender's Key" during the MPTCP Support Check, identifes the MPTCP Fullmesh, displays MPTCP monitor data and Established connections consistent with a working system on the "Network-MPTCP" page.  As it is setup consistent with the "Beta" system outlined above, all the services are working appropriately except outbound SMTP.  That being said, I recall having an identical problem developing the original "Beta" version, but don't recall at the moment how I resolved that issue.  It's related to a MPTCP inability or misidentification of the receiving server so for all intents and purposes, I believe this deployment is functioning quite consistently with the "Beta" deployment.

Finally, openmptcprouter-v0.59.1-5.15-r0+20029-3c06a344e9-x86-64-generic-ext4-combined.img was installed on the same hardware and services as identified below with a VPS using Debian 11 and the 0.1028 script.  Upon completing the setup, the "Status" and "Dashboard" functioned accordingly.  This version would be desired as we move forward, however, the T-Mobile and Starlink services both fail the "MPTCP Support Check" of the "Network" tab while succesfully providing a connection to the internet as intended.  Using the 5.4 kernel on the same exact setup successfully acquires a "Sender's Key" during the check.

## Steps to Reproduce the Problem
Simply complete the install per the "[Router install](https://github.com/Ysurac/openmptcprouter/wiki/Router-install)" documentation at OpenMPTCProuter.com

## Context (Environment)
Trying to improve our deployment by upgrade from beta software and taking advantage of core improvements anticpated in the updated version of ModemManger as our 5G modem is very new technology and will likely have better more reliable performance with the latest software.  We expect to deploy this solution once stabilized on a larger scale and wish to continue our testing and feedback contributions to aide in the projects maturation.

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8
  - OpenMPTCProuter VPS version: Debian 11 5.4.207-mptcp VPS 0.1028
  - OpenMPTCProuter VPS provider:  IONOS.com 2 vCore 2GB RAM "Type M VPS"
  - OpenMPTCProuter platform: [Lenovo 5100C1U SFF](https://pcsupport.lenovo.com/us/en/products/desktops-and-all-in-ones/thinkcentre-m-series-desktops/thinkcentre-m78/5100/document-userguide/doc_userguide) with an [AMD A4-5300 APU](https://www.amd.com/en/support/apu/amd-series-processors/amd-a4-series-apu-for-desktops/a4-5300-radeon-hd-7480d) and one additional Intel NIC adapter.
  - Country: United States

Ysurac, 2022-09-15:
If you have "no data" it's because some checks done via status page doesn't doesn't answer. You may have some info in Status->System log.
For 5.15 based release, it's still a test. I didn't tested if the Multipath TCP check was still working on this release.

Network-Traditions, 2022-09-15:
Good to know about the 5.15 kernel.  I do recall also connecting to a [Multipath TCP Check online](http://amiusingmptcp.de/) and getting a no MPTCP result if this feedback is helpful.  I will look deeper into the System log for clues as well as compare the working systems to the no working one to flush out differences.  Any guidance on errors, warnings, notices, sections, devices, services etc to be looking at for my issue?

Ysurac, 2022-09-15:
This website is not updated to use MPTCPv1, it only work for MPTCPv0 used in OpenMPTCProuter based on 5.4 kernel.

Network-Traditions, 2022-09-15:
Also good to know.  Rebuilt the system from scratch.  Upon updating the configuration to connect for the first time, the Dashboard and Network Overview pages were propagating appropriately.  While I had my reasons for the organization of my nomenclature and device assignments, as I attempted to modify the eth0/LAN adapter assignment and naming to a WAN device, the issue reported here returned.  Reassigning the eth0 as a LAN named device solved this issue.  Was having difficutly re-establishing an MPTCP connection, however, a VPS reboot solved that issue.  All seems to be going well so we are looking forward to getting this system fully configured to document the improvements in performance and reliability as compared to our Beta6 version.  Thanks again!!!

closed 2022-09-15 by Network-Traditions

Network-Traditions, 2022-09-16:
Not sure if these System Log entries were related to this problem/resolution, but in the event others encounter this or related issues here are the entries I believe were generated as a result of the improper configuration:

<warn>  [SQM_IFB_77e2d] invalid sysfs path read for net/SQM_IFB_77e2d
<warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_77e2d not found
<warn>  [SQM_IFB_490eb] invalid sysfs path read for net/SQM_IFB_490eb
<warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_490eb not found

Ysurac, 2022-09-16:
Not related, not a problem. Related to SQM virtual interfaces.