Ysurac/openmptcprouter #2541: OMR-Bypass not bypassing domains, traffic still going out through all wan interfaces.

ioogithub, 2022-09-15:
## Expected Behavior
When a domain name is entered into the Servies>OMR-Bypass screen, traffic to that website will bypass the aggregate OMR tunnel and only go out over one wan connection, like a pass-though bypassing the OMR bonded routing.

## Current Behavior
speedtest.net and netflix.com traffic still goes out though the OMR tunnel and traffic is seen on both tunnels in the Network->Interfaces->Bandwidth screen.

## Possible Solution

Fix the software so when a domain name is entered into this screen, the traffic will bypass OMR and pass-though directly to the wan selected only.

## Steps to Reproduce the Problem
  1. Services->OMR-Bypass-> under 'Domain' add a domain "speedtest.net" -> Save
  2. Open a browser window, enter "speedtest.net" and start the test
  3. Network->Interfaces->Bandwidth->Observe that traffic saturates both wan connections.
  4. Repeat step 1 and for Interface select "eth1"->Save
  5. Repeat step 2 and 3. same result.
  6.  Repeat step 1 and for Interface select "eth2"->Save
  7. Repeat step 2 and 3. same result.
  8. Repeat steps 1-3 with another domain "netflix.com" and observe the same results. Traffic always goes out over both wan lines, it is not bypassed.
So it doesn't seem to matter that is selected for interface the traffic always goes out though the OMR tunnel. It is not bypassed to one wan connection as expected.

## Context (Environment)

I am trying to have certain websites such as netflix.com bypass the OMR tunnel and only go out over one wan interface.

Also, the contect hint under the Interface dropdown needs clarification. If this the interface we are tryingto bypass or the interface we wish the traffic to go out on. Perhaps change the working from:

`When none selected, MPTCP master interface is used.`

`to

`Select the interface where the traffic should be directed. When none selected, MPTCP master interface is used`

to make things more clear.

Either way I tested with both interfaces and the traffic always goes out though both of them regardless of what I enter for itnerface so it doesn't appear to be working.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.59.1-5.4 r0+16594-ce92d
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider: linode
  - OpenMPTCProuter platform: RPI4B

## Supporting Documentation

I tried each domain with interface: eth1, eth2 and default and the results were always traffic going out to both wan lines:
[image]

Here are the results of the tests. I have two wan interfaces: eth1, eth2 and default MPTCP (set to eth2):
[image]
[image]
[image]

Ysurac, 2022-09-15:
When you set speedtest.net or netflix.com, if you set correctly your DNS to use OpenMPTCProuter IP, it bypass *.speedtest.net and *.netflix.com, but not all domains these websites can use.
Netflix use a domain like nflxvideo.net to display video and some CDN. It's the same for speedtest, to contact a speedtest server it use it's domain name like mytestserver.provider.com

ioogithub, 2022-09-15:
> When you set speedtest.net or netflix.com, if you set correctly your DNS to use OpenMPTCProuter IP, it bypass *.speedtest.net and *.netflix.com, but not all domains these websites can use.

Okay I will try to gather all domain that these website use and add them in the Domains screen. If anyone else in the future reads this, 'dnstop' is a really nice and quick way to see all host names a website is requesting.

Can you explain the "set DNS to use OpenMPTCProuter" more clearly?

Currently I have this DNS configuration:
`client->mesh_router->omr->VPS
`
1. client:
```
cat resolv.conf
# Generated by NetworkManager
nameserver 127.0.0.53
```

2. mesh_router:
`127.0.0.1#53

3. OMR (defaults)
[image]

4. vps:
`nameservers are here`

How do I have to configure DNS to get this working?
```

Ysurac, 2022-09-15:
OMR use DNS root servers by default (so it don't care about VPS DNS). mesh router should relay to OMR IP, and client to OMR IP directly if possible or mesh router IP if it's a DNS relay.

ioogithub, 2022-09-15:
> OMR use DNS root servers by default (so it don't care about VPS DNS)
So leave this as it is, default behavior?

> mesh router should relay to OMR IP
Mesh router is also OpenWRT, are you saying to do this?
[image]

> client to OMR IP directly
This is the hardest part fighting with Network Manager, it always overrides the settings and want to keep it at 127.0.0.53. Do you think it will work if I make the change to the mesh router or do I absolutely have to set each client manually with OMR IP for DNS?

ioogithub, 2022-09-16:
I think I have it working now, thank you for the advice.

Feature request: wildcard characters for the blocking entries would really solve some problems with this feature.

Ysurac, 2022-09-16:
wildcard are allowed on subdomain and it's always a wildcard. If you put netflix.com, this will bypass *.netflix.com too, so www.netflix.com, toto.netflix.com,...

ioogithub, 2022-09-17:
Does the bypass work with omr-test-speed? I am trying to bypass one wan connection and force all traffic though the other one using a bypass but traffic always goes out though both wans.

I tried the following:

Domain:
datapacket.com
download.datapacket.com
nyc.download.datapacket.com

Ports:
80
8080

none of these work, the traffic always goes out though both.

Ysurac, 2022-09-17:
```omr-test-speed wan1``` will test speed using physical interface wan1.
omr-bypass, bypass the VPS, so interface is the output traffic interface.

ioogithub, 2022-09-17:
Is this the same result then:

1. `omr-test-speed wan1` (starlink) = traffic bypasses VPS and goes though wan1 directly
2. `ome-test-speed` with `omr-bypass` rule` nyc.download.datapacket.com wan1` = traffic bypasses VPS and goes through wan1 directly.

I can't get #2 to work, the bandwidth graph still shows the omr-test-speed traffic going out though both wan1 and wan2.

Does the omr-test-speed script look at the omr-bypass rules or is omr-bypass only for a client with a web browser?

Ysurac, 2022-09-18:
omr-bypass is for client computer. And I don't really know why bypass omr-test-speed when it's not needed.

closed 2022-12-22 by github-actions[bot]