Ysurac/openmptcprouter #2542: CalAMP Fusion

squeaky-geek, 2022-09-16:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

      
not necessarily, more of a device compatibility check, the SoC in question is x86

**Describe why you are asking for this support?**

these boards are a perfect candidate for 4g lte bonding

**Additional context**

please see images

 " CalAmp Fusion lte modem, RE BRANDED as a " Harris MBC-200 " 
it boots " busybox " on a MPC8314E PowerQUICC II Pro Processor... ( see images ) -- some logs / more intel available on request --
My question(s) : how do i go about re flashing this? ( without bricking )
-- for that matter... is it even re flashable?
-- the firmware files from CalAmp are " .pak " files How do i edit / view / dissect them?
-- the nand is a SPANSION GL512N11FFI02
-- it has one visible ddr2 chip that i think is either 512MB or more likely 128Mb..

any help / insight is greatly appreciated, BOTH calamp AND harris ( go figure ) have been of no help. ( in their defense " yet..." )

my end goal is to run dual concurrent 4g modems with Openmptcprouter.
[image]
[image]

Ysurac, 2022-09-16:
Seems to be a PowerPC, so not x86 compatible.
The processor seems to be slow.
This seems really difficult to support this board as I don't see any OpenWRT/Linux support.

squeaky-geek, 2022-09-16:
Thank you for your response. I'll put it on a shelf in hopes of future
Linux support.

squeaky-geek, 2022-09-17:
@Ysurac , if i post the log that i managed to get do you think you may be able to point me in the direction of dissecting this thing?
  i really hate to scrap it, if i have to compile / roll my own firmware i'm not opposed. ( even if its not linux per say )

squeaky-geek, 2022-09-17:
Jan  1 00:00:05 ltib syslog.info syslogd started: BusyBox v1.21.1
Jan  1 00:00:05 ltib user.notice kernel: klogd started: BusyBox 

v1.21.1 ()
Jan  1 00:00:05 ltib user.info kernel: Device-tree (9300100-PLT0) 

matches with the actual board model (9300100-PLT0).
Jan  1 00:00:05 ltib user.info kernel: Using DR 9300100 machine 

description
Jan  1 00:00:05 ltib user.notice kernel: Linux version 2.6.29.6-10-

dr1 (dataradio@lucee-vm) (gcc version 4.1.2) #1 Mon Mar 9 19:19:24 

EDT 2015
Jan  1 00:00:05 ltib user.debug kernel: Found legacy serial port 0 

for /immr@e0000000/serial@4600
Jan  1 00:00:05 ltib user.debug kernel:   mem=e0004600, 

taddr=e0004600, irq=0, clk=132665334, speed=0
Jan  1 00:00:05 ltib user.debug kernel: Found legacy serial port 1 

for /immr@e0000000/serial@4500
Jan  1 00:00:05 ltib user.debug kernel:   mem=e0004500, 

taddr=e0004500, irq=0, clk=132665334, speed=0
Jan  1 00:00:05 ltib user.info kernel: console [udbg0] enabled
Jan  1 00:00:05 ltib user.info kernel: Found FSL PCI host bridge at 

0x00000000e0008500. Firmware bus number: 0->0
Jan  1 00:00:05 ltib user.info kernel: PCI host bridge /pci@e0008500 

(primary) ranges:
Jan  1 00:00:05 ltib user.info kernel:  MEM 

0x0000000090000000..0x000000009fffffff -> 0x0000000090000000 
Jan  1 00:00:05 ltib user.info kernel:  MEM 

0x0000000080000000..0x000000008fffffff -> 0x0000000080000000 

Prefetch
Jan  1 00:00:05 ltib user.info kernel:   IO 

0x00000000e0300000..0x00000000e03fffff -> 0x0000000000000000
Jan  1 00:00:05 ltib user.info kernel: Found FSL PCI host bridge at 

0x00000000e0009000. Firmware bus number: 0->0
Jan  1 00:00:05 ltib user.info kernel: PCI host bridge /pci@e0009000 

 ranges:
Jan  1 00:00:05 ltib user.info kernel:  MEM 

0x00000000a0000000..0x00000000afffffff -> 0x00000000a0000000 
Jan  1 00:00:05 ltib user.info kernel:   IO 

0x00000000b1000000..0x00000000b17fffff -> 0x0000000000000000
Jan  1 00:00:05 ltib user.info kernel: Found FSL PCI host bridge at 

0x00000000e000a000. Firmware bus number: 0->0
Jan  1 00:00:05 ltib user.info kernel: PCI host bridge /pci@e000a000 

 ranges:
Jan  1 00:00:05 ltib user.info kernel:  MEM 

0x00000000c0000000..0x00000000cfffffff -> 0x00000000c0000000 
Jan  1 00:00:05 ltib user.info kernel:   IO 

0x00000000d1000000..0x00000000d17fffff -> 0x0000000000000000
Jan  1 00:00:05 ltib user.debug kernel: Top of RAM: 0x4000000, Total 
[... 1460 lines cut ...]
Jan  1 00:01:19 MBC-200 daemon.info udhcpd[1701]: udhcpd (v1.21.1) 

started
Jan  1 00:01:19 MBC-200 daemon.err udhcpd[1701]: can't open 

'/var/udhcpd-wlan0.leases': No such file or directory
Jan  1 00:01:19 MBC-200 daemon.err udhcpd[1701]: is interface wlan0 

up and configured?: No such device
Jan  1 02:10:54 MBC-200 daemon.info udhcpd[1317]: Sending OFFER of 

192.168.1.120
Jan  1 02:10:54 MBC-200 daemon.info udhcpd[1317]: Sending OFFER of 

192.168.1.120
Jan  1 02:10:54 MBC-200 daemon.info udhcpd[1317]: Sending ACK to 

192.168.1.120
Jan  1 02:14:31 MBC-200 user.notice root: IN pre_uImage ...
Jan  1 02:14:32 MBC-200 user.notice root: 

expected_md5:7e9c17eb5b302b7dc2fc2e4f2e8005ce 

actual_md5:b8060e17539be179d07f76a78066bc73
Jan  1 02:14:32 MBC-200 user.notice root: Erasing uImage

squeaky-geek, 2022-09-21:
is there a 32 bit ( i386 / x86 ) server image? or will the generic 32 / 64 image work? trying to roll my own server with a stratacache spectra s-200 and debian 11.05 i386. ( or will i need to compile from source? ( ive never compiled from source before so that will be fun if needed ) -thanks for putting up with me xD

Ysurac, 2022-09-22:
Only x86_64 is supported server side.
You would need a x86 patched compiled kernel and use i386 packages if available. Good luck...

closed 2022-12-26 by github-actions[bot]