Ysurac/openmptcprouter #2544: Intel i226-v support on x86_64?

ahayes, 2022-09-17:
## Expected Behavior

After installing openmptcprouter on a new ChangWang N6005 miniPC with 4 x intel i226-v ports, I would expect to be able to connect to it over ethernet using SSH or HTTP at the 192.168.100.1 address in the documentation.

## Current Behavior

I am unable to connect to the system. I think perhaps the i226-v ethernet is not recognized and supported.

## Specifications

  - OpenMPTCProuter version: [openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined-efi](https://download.openmptcprouter.com/release/v0.59.1/x86_64/targets/x86/64/openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined-efi.img.gz)
  - OpenMPTCProuter VPS version: Not applicable yet.
  - OpenMPTCProuter VPS provider: Not applicable yet.
  - OpenMPTCProuter platform: x86_64

Ubuntu 22.04 live iso recognized and was able to use the ethernet adapters so I know they work. Should I try your [OpenMPTCProuter based on kernel 5.15 using upstream MPTCP](https://download.openmptcprouter.com/release/v0.59.1-5.15/) build with the newer linux kernel? Is there a way to run openmptcprouter on linux on the router end as well as the VPS end instead of the openwrt base?

Ysurac, 2022-09-17:
Should work with 5.15 based release.

ahayes, 2022-09-17:
Thank you. I will try that this weekend.

ahayes, 2022-09-17:
The Intel i226-v interfaces are recognized and working with [openmptcprouter-v0.59.1-5.15-r0+20029-3c06a344e9-x86-64-generic-ext4-combined-efi.img](https://download.openmptcprouter.com/release/v0.59.1-5.15/x86_64/targets/x86/64/openmptcprouter-v0.59.1-5.15-r0+20029-3c06a344e9-x86-64-generic-ext4-combined-efi.img.gz)

I have it up inside my existing network for testing with one static WAN configured. When I tried DHCP with that WAN it just kept coming up, grabbing an IP, then resetting. Once I set it to static it stopped resetting. I had to set MPTCP over VPN to make it happy, but as I type this I am behind my new openmptcprouter box so I'm excited. Thank you for your help! I will close this and open a new issue if I have any more hiccups putting the router in place for real.

closed 2022-09-17 by ahayes

ioogithub, 2022-09-20:
@Ysurac, you mentioned that the 5.15 kernel:

Does the Intel i225-v NIC also require the 5.15 kernel? There is an issue here (https://github.com/openwrt/openwrt/issues/9403) and it was recommended to use the 5.15 kernel.

If it is required, what limitations are there? It says:
"Vvailable for tests, have some limitations)."

Is this kernel production ready, what are the limitations?