Ysurac/openmptcprouter #2547: USB LTE 4G modem is recognized as Mass Storage

purpelhorse, 2022-09-17:
Have two USB dongles: Alcatel IK41VE1 and Huawei E8372H-153
I managed to configure Alcatel properly (have internet access).

Same steps but just for one E8372 failed as the modem is not even recognizable (usb0 is not shown as an option) and the console says it is a "Mass storage". No issues on Windows though. 

I found that such issues (USB to cellurar device) happen on Ubuntu for some devices, and the solution is libusb/usb-modeswitch, but so far failed miserably and don't know if it will eventually work. Any help/ideas please?

Ysurac, 2022-09-18:
It's not in Hilink mode and give a eth* interface ?
You can check on OpenWRT forum if you can have any info on this modem.

purpelhorse, 2022-09-20:
Looks like I managed to switch the modem from USB Mass Storage to a cellular device via usb-mode.json (now the console shows this Huawei modem as a cellular device)

"Physical interface" also switched from eth* to wwan0 (is this ok?)

Now my test settings are:
Interface: wan1
Type=MacVLAN
Physical interface=wwan0 (other options are eth0 and wlan0)
IPv4 address=192.168.30.97 (any random ip?)
IPv4 gateway=192.168.3.1 (ip of the modem, DHCP is disabled on the modem)

It worked fine for my other modem (Alcatel), but now I am having "Gateway DOWN" error (crossed circle) and no signs that the modem is somehow identified on Network Overview. 

Am I missing something again? Please help.

Ysurac, 2022-09-20:
IPv4 address of the interface must be in same subnet as IPv4 gateway. As subnet if often 255.255.255.0 (/24), put in address in the range 192.168.3.2-192.168.3.254

closed 2022-12-25 by github-actions[bot]