Ysurac/openmptcprouter #2552: OMR on RPi 4b using LAN switch instead of USB LAN adapters for line bonding

L1TTL3big, 2022-09-20:
I'm currently using OMR on RPI 4b in such a way that LTE/5G modems are connected to RPi via USB(3.0) LAN adapters and another router (yeah might be strange but LAN setup is inherited before I was aware of OMR - the whole network was already set up and I'm not keen to reassemble it) is connected to RPi LAN port. Somehow I ended up using one of such USB LAN adapters on some of my PCs and I've noticed they're actually decreasing latency/speed opposed to "classical" LAN cards.
Therefore I'd like to know if it would be possible to attach both LTE/5G modems and router on LAN switch (which can be configured to work as both managed and unmanaged switch) and then connect switch to RPi LAN port instead of using USB ports. I'm quite sure this way I would gain some latency improvement over using USB LAN adapters. I'm LAN basics aware but I'd like to know if (i.e. #2003) this setup would work and if someone could provide me some basic instructions how to implement it. I guess by just connecting everything to switch and then switch to RPi wouldn't work out of the box ...
Thank you - much appreciated.

PS.
I've posted this same topic in Discussions (https://github.com/Ysurac/openmptcprouter/discussions/2538) but with no answer, so maybe someone will be able to help me out here ...

Ysurac, 2022-09-20:
By default RPI configuration is made to have everything on same switch without using any USB to LAN adapter.
For that MacVLAN is used, and you only need to follow the documentation for the configuration.

L1TTL3big, 2022-09-20:
Wow, that was fast ... Hmm, I'm no networking guru ... Could you please be a little bit more specific ... Thank you ...

Ysurac, 2022-09-20:
Yes, you then need to configure VPS key and WANs IP in the wizard.

L1TTL3big, 2022-09-20:
Great! Thank you. Will try as soon as everybody goes to sleep as this is my only WAN connection ...
VPS addresses and keys are already in place as I'm using OMR with USB adapters already ... WAN IP's are set to DHCP for (my) convenience ... Hopefully I'll work ...

Ysurac, 2022-09-20:
If you put all on same switch you need to set all WANs and modems with static IPs.

Ysurac, 2022-09-20:
And do a configuration backup before testing changes.

L1TTL3big, 2022-09-20:
OK. Got it working ... But I must say I'm pretty confused ... How is OMR working with static IP addresses of 5G modems when wan1 and wan2 interfaces are changing their MAC addresses with each interface restart or device reboot ... and when I can't actually see OMR connected to one of those routers (it doesn't show up in DHCP server list at all) and the other is catching up current wan MAC address but since I've assigned it as static I'm quite curious what will happen on next OMR reboot ... Huh ... It just added another OMR with another MAC on the list ... obviously is this about some kind of MacVLAN magic ... Really don't get it how it's all working ...

lkimloong, 2022-10-11:
I configure the same setup as @L1TTL3big and i run with an issue. My setup is using RPi3b+ connected to a 100mb 5 port switch with 3 4G router. Currently is only using 2 sim card. I am not able to get into the OMR web ui whenever the sim card is inserted. Only able to get in the OMR web ui when i eject each of the sim. Is kind of strange. If I try to log in to the 4G router it behave the same until i have to disconnect OMR and other 4G router from the 5 port lan switch. I did try to change the switch and the behavior is the same. Appreciate if i can get some assistance from here. Sorry for my bad english.

Ysurac, 2022-10-11:
I think you have some network configuration problem. Check that you don't have any duplicate IP.

closed 2023-01-14 by github-actions[bot]