Ysurac/openmptcprouter #2555: When can I support the RM500U 5G module driver for Mobile?

eirdgms, 2022-09-20:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

**Describe why you are asking for this support?**

**Additional context**

eirdgms, 2022-09-21:
I want to use the Modena Manager to get the network

suyuan168, 2022-09-21:
RM500U is suitable for mbim support usb net network card. But modemmanager is not supported

eirdgms, 2022-09-22:
大魔王

eirdgms, 2022-09-22:
Recently i bought a new 5G modem (Quectel RM-520N-GL), but both drivers/net/usb/qmi_wwan.c and drivers/usb/serial/option.c aren´t prepared for this new modem.

suyuan168, 2022-09-22:
> Recently i bought a new 5G modem (Quectel RM-500U-CN), but both drivers/net/usb/qmi_wwan.c and drivers/usb/serial/option.c aren´t prepared for this new modem.

hi：  

As far as I know rm500u is made in China and it is mbim compliant.
VID: 0x2c7c
PID: 0x0900
Need to add a usb driver.
It only supports USB RNDIS/ECM/NCM/MBIM network cards.
Modemmanager should support mbim dialing, and you can also make it work by adding mbim.

Module's VID and PID Linux USB Driver Interface
VID: 0x2c7c
PID: 0x0900
USB RNDIS/ECM/NCM/MBIM network card Interface 0/1: USB network adapter
USB to Serial option
Interface 2: DIAG command communication port
Port 3: LOG port
Port 4: AT command communication port
Interface 5: Modem command communication port
Port 6: NMEA command communication port
USBFS interface 7: ADB command communication port

1,
Add the module's VID and PID information in the file [KERNEL]/drivers/usb/serial/option.c, for example:
static const struct usb_device_id option_ids[] = {
#if 1 //Added by Quectel
{ USB_DEVICE_AND_INTERFACE_INFO(0x2c7c, 0x0900, 0xff, 0x00, 0x00) },
#endif
Important: Module interface 2/3/4/5/6 are serial ports, and it is necessary to ensure that the number of USB ports exceeds 4.
After the interface will not be filtered out
2,
Please add the following statement to the file
[KERNEL]/drivers/usb/serial/usb_wwan.c
static struct urb *usb_wwan_setup_urb(struct usb_serial *serial, int endpoint,
 int dir, void *ctx, char *buf, int len, void (*callback) (struct urb *))
{…
usb_fill_bulk_urb(urb, serial->dev,
 usb_sndbulkpipe(serial->dev, endpoint) | dir,
 buf, len, callback, ctx);
#if 1 //Added by Quectel for zero packet
if (dir == USB_DIR_OUT) {
struct usb_device_descriptor *desc = &serial->dev->descriptor;
if (desc->idVendor == cpu_to_le16(0x2C7C))
urb->transfer_flags |= URB_ZERO_PACKET;
}
#endif
return urb

3.
The module supports MBIM/RNDIS/ECM/NCM four network card functions, Linux system supports these USB network card functions by default, the system is built-in
The driver module does not require any modification to the driver files of the Linux system. The driver source code is maintained by GNU Linux.
After the module is connected to the Linux Host and successfully loaded with the corresponding network card driver for USB, a network card will be generated on the Host. cdc_mbim
A cdc-wdm character device will also be generated for command interaction, so it is more appropriate to control AT through minicom.
ECM cdc_ether configure ECM mode via AT+QCFG="usbnet",1
MBIM
cdc_mbim
(Kernel 3.18 and above)
Configure MBIM mode via AT+QCFG="usbnet",2
RNDIS rndis_host configure RNDIS mode via AT+QCFG="usbnet",3
NCM cdc_ncm configure NCM mode by AT+QCFG="usbnet",5.

Here's some information for you:
1. In network card mode and router mode, if both USB and Ethernet network cards are connected at the same time, only one of the network cards can obtain an IP address. two
When the network cards are enabled at the same time, the Ethernet network card has priority to obtain the IP address.
2. In bridge mode, if both USB and Ethernet network cards are connected at the same time, both network cards can obtain IP addresses.
3. In network card mode, the IP address obtained by the host computer is the IP address obtained by the core network, and the module cannot connect to the external network.
4. In bridge mode and routing mode, the IP address obtained by the host computer is the IP address assigned by the internal LAN of the module.
Connect to the external network.
First, configure the network card through AT+QCFG; secondly,
Dial-up Internet access via AT+QNETDEVCTL. After successful dialing, you can check the status of the network card through AT+QNETDEVSTATUS.
Windows system
Execute AT+QCFG="usbnet",3 to configure the dial-up mode of the network card and the driver type as RNDIS;
Or execute AT+QCFG="usbnet",2 to configure the network card dial-up mode and driver type as MBIM, and you can also use mbim dial-up and modemmanager dial-up in OMR, but it is not recommended, it is a waste of system resources.
Linux system
Execute AT+QCFG="usbnet",5 to configure the dial-up mode of the network card and the driver type as NCM;
Or execute AT+QCFG="usbnet",1 to configure the dial-up mode of the network card and the driver type as ECM;
Or execute AT+QCFG="usbnet", 2 to configure the dial-up mode of the network card and the driver type as MBIM;
Or execute AT+QCFG="usbnet",3 to configure the dial-up mode of the network card and the driver type as RNDIS.
Configure dial mode
Execute AT+QCFG="nat",0 to configure the dial-up mode as network card mode;
Or execute AT+QCFG="nat",1 to configure the dialing mode as routing mode;
Or execute AT+QCFG="nat",2 to configure the dial-up mode as bridge mode.

Configure the Ethernet network card
RG200U-CN, RG500U-CN and RM500U-CN modules support enabling or disabling the Ethernet network card through AT commands, and the default
Confirm to open.
Execute AT+QCFG="ethernet",1 to turn on the Ethernet network card;
Or execute AT+QCFG="ethernet",0 to turn off the Ethernet network card.

How to use it depends on the method that suits you. It is recommended to use
@Ysurac buddy,Please add it to the driver patch if you have time.

eirdgms, 2022-09-22:
谢谢

psfabio, 2022-09-23:
Hi suyuan168,
i did notice that you know a lot about this modules.
if you please could help me..
I´m using a EM12G in qmi mode (qmi mode on openmptcprouter (i think they use uqmi to connect), and qmicli to get status)
Recently i bought a RM520N module and thanks to Ysurac patch i can access it via qmi too, but both uqmi and qmicli seems a bit outdated for 5g modules.
What do you think is the best way to both connect and get status (operator, signal, etc) on mptcp router.
I don´t like the modemmanager much, because i think it do a mess with the interfaces names, etc.
I was reading about RGMII, but there is few documentation about it.

thanks a lot.

suyuan168, 2022-09-23:
> Hi @suyuan168 , i did notice that you know a lot about this modules. if you please could help me.. I´m using a EM12G in qmi mode (qmi mode on openmptcprouter (i think they use uqmi to connect), and [...]

@Ysurac Buddy default integration kit minicom.
For modemmanager if you use this as a dialer tool.
1. Then you should disable the network card with the same name in the advanced settings page of the setup wizard.
2. When you add modemmanager to the interface, if there are multiple RM520Ns, you can choose different RM520Ns. This will not be confusing, as long as each interface selection is not repeated, you can test your SIM cards one by one. This can find the corresponding RM520N.
3. For QMI, it is the standard dial-up interface of Qualcomm RM520N, as well as gobinet. No effect on speed.
4. It is recommended to use modemmanager for interface management.
5. For the interface of RM520N, I can give you the following information:
If you use modemmanager you can use the command:
mmcli -L lists all current RM520Ns.
mmcli -m 0 can enumerate the detailed RM520N status, including the USB interface.
In this way, you can set the corresponding modemmanager on the interface details page.
For example mmcli -m 0 can see the current at port.
For example, the current module 1 corresponds to the at port USB2.
6. Whether you use qmi or modemmanager, you can ls /dev to view the current USB port.
7. Command minicom -s /dev/ttyUSB2 Suppose this is the interface at of the first module of RM520.
8. You should google about minicom.
You should enter the at state and type AT&F for command echo.
AT+CSQ to check the current network status of the SIM card or module.
AT+CFUN=0 close the module
AT+CFUN=1 open the module
AT+CPIN? Check SIM card readiness
AT+COPS? View signal status

Tips You should learn to debug one by one to find your corresponding SIM card and module. This should be simple.

psfabio, 2022-09-23:
thanks for the tips..
Could i use the modemmanager and issue at commands via minicom at same time?

suyuan168, 2022-09-23:
> thanks for the tips.. Could i use the modemmanager and issue at commands via minicom at same time?

yes！why not try it

eirdgms, 2022-09-23:
[image]

closed 2022-12-27 by github-actions[bot]

xingminghan, 2023-04-25:
hello, i cannot find this file "[KERNEL]/drivers/usb/serial/option.c". do you know how to find it?

avatar21, 2024-10-11:
Does anyone by any chance tried this 5G module with Malaysia telcos' SIM card? Does it work?