Ysurac/openmptcprouter #2556: Question about IPv6

obi12341, 2022-09-21:
When ipv6 is enabled all devices get an ULA ipv6 address. Is it possible, if the VPS has a /64 network attached, to hand out the Clients a GUA address?
And how is it working with just the ULA address, is it using NAT64?

Thanks in advanced. And some love for this awesome project <3

Ysurac, 2022-09-22:
Better with a /56, but may work if you don't use already used VPS IP.
This can be set in the wizard or in Network->Interfaces menu.

obi12341, 2022-09-25:
Hmm not sure if I understand correctly, but the IPv6 Prefix has to be assigned to the VPS. So in Network->Interface-> OMR6IN4 i have to put it my IPv6 Prefix i got from my VPS Provider in "IPv6 routed prefix" or somewhere else? In the LAN interface probably not because this has only the ULA Prefix set and probably is not able to set up routing, or am I wrong? :P 
Thanks in advanced.

Ysurac, 2022-09-25:
In System->OpenMPTCProuter, Wizard tab, "advanced settings" checbox, "IPv6 prefix"

obi12341, 2022-09-25:
Ah there it is. Strange thing about it, if I set this prefix it gets reseted to the ULA prefix. Is it a bug or is something wrong on my side? I have ordered a /56 now and tried also with it, but same problem.
```2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether f2:3c:93:f9:41:0a brd ff:ff:ff:ff:ff:ff
    inet 192.46.xxx.xxx/24 brd 192.46.xxx.xxx scope global eth0
       valid_lft forever preferred_lft forever
    inet6 2a01:7e01:e002:xxx::/56 scope global
       valid_lft forever preferred_lft forever
    inet6 2a01:7e01::xxx:93ff:fef9:410a/64 scope global dynamic mngtmpaddr
       valid_lft 5367sec preferred_lft 1767sec
    inet6 fe80::f03c:xxx:fef9:410a/64 scope link
       valid_lft forever preferred_lft forever
```
this is my network config on the VPS and i try to set up the 2a01:7e01:e002:xxx::/56 in the wizard.

Ysurac, 2022-09-30:
Check in Network->Interfaces, "Global option" that IPv6 Prefix is correct.

closed 2023-01-04 by github-actions[bot]