Ysurac/openmptcprouter #2562: GUI stops working randomly and can only be recovered with a reboot

tlangley93, 2022-09-22:
## Expected Behavior

When going to the IP address of OMR the WebUI normally loads to the logon screen

## Current Behavior

When going to the IP address of OMR this error message appears

/usr/lib/lua/luci/dispatcher.lua:427: /etc/config/luci seems to be corrupt, unable to find section 'main'

## Possible Solution

Only solution that I have found is to restart the router from SSH to recover the UI

## Steps to Reproduce the Problem

  1. Make a settings change EX: Change the VPS server that OMR points to
  2. Attempt to access another part of the UI
  3. Error message appears

## Context (Environment)

I cannot check the status of or change any further settings without a reboot which causes the internet to drop out during the restart. I appreciate the work you are doing on this and although I am not a developer I am a network engineer so I will try and assist in any way I can.

## Specifications

  - OpenMPTCProuter version:  0.59.1-5.4
  - OpenMPTCProuter VPS version:  0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Was Vultr now Linode
  - OpenMPTCProuter platform:  X86_64
  - Country:  United States

Ysurac, 2022-09-22:
It's often a crash of rpcd service. Check Status->System log if there is any info about that.
Check also that "uci show" doesn't give any error via ssh on the router.

myshophosting, 2022-10-19:
I have had this issue through several versions and it continues in the latest 0.59.1-5.4 - my router is x86.

The UI is almost unusable. I cannot see anything about rpcd in the System Log either.

Ysurac, 2022-10-19:
Try to restart rpcd service.

myshophosting, 2022-10-19:
Actually I have reread the original issue reported, my issue is different. I don’t get that error at the login screen - but my UI is very much unusable. It’s extremely slow at times, and pages don’t load. A reboot seems to fix it sometimes for a short while, but browsing around in the UI is painful.