Ysurac/openmptcprouter #2563: V2RAY Frequently Stop

ccmks, 2022-09-22:
OMR Version 0.59.1-5.4
VPS Version: 1028

Need to switch back to Shadowsocks then switch back to V2RAY in order to fix it. But it won't last long, it will break again!

The log will be provided on next thread

Please help, thank you very much!

ccmks, 2022-09-22:
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.802993] [  24403]   123 24403      289        8    32768        0             0 ntpd
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.805316] [  28057] 65536 28057     3605     1695    73728        0             0 unbound
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.808308] [  28190]     0 28190     1271       92    45056        0             0 ttyd
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.811124] [  26777]     0 26777      353       77    45056        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.815043] [  26778]     0 26778      352       77    40960        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.817453] [  26779]     0 26779      352       76    40960        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.819842] [  26780]     0 26780      352       77    36864        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.822338] [  26781]     0 26781      339       62    40960        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.825135] [  26782]     0 26782      294       17    40960        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.827732] [  12547]     0 12547  1333476   109735  1003520        0             0 v2ray
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.830012] [  20699]     0 20699      257       33    36864        0             0 luci-bwc
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.832670] [   4599]     0  4599      463      188    40960        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.834994] [   5408]     0  5408      320       44    40960        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.836887] [   5425]     0  5425      444      169    36864        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.839335] [   5648]     0  5648      320       43    40960        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.841150] [   5651]     0  5651      299       10    40960        0             0 xargs
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.842874] [   5652]     0  5652      291       10    36864        0             0 tr
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.844790] [   5658]     0  5658     3782      191    61440        0             0 dig
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.846627] [   5707]     0  5707      422      144    40960        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.848145] [   5717]     0  5717      931       38    36864        0             0 curl
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.850136] [   5731]     0  5731      436      119    45056        0             0 luci
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.852943] [   5782]     0  5782      422      144    40960        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.854425] [   5784]     0  5784      460       12    32768        0             0 ip
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.855889] [   5785]     0  5785      289        8    32768        0             0 grep
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.858274] [   5786]     0  5786      291        7    40960        0             0 awk
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.859601] [   5787]     0  5787      352       76    40960        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.863870] [   5789]     0  5789      353       77    45056        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.867056] [   5795]     0  5795      444      169    36864        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.868212] [   5798]     0  5798      291       10    36864        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.869348] [   5799]     0  5799      215       12    40960        0             0 jsonfilter
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.870459] [   5801]     0  5801      352       76    40960        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.871570] [   5802]     0  5802      353       77    45056        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.874118] [   5803]     0  5803      291       13    36864        0             0 sh
Thu Sep 22 20:40:01 2022 kern.info kernel: [ 2674.877748] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=v2ray,pid=12547,uid=0
Thu Sep 22 20:40:01 2022 kern.err kernel: [ 2674.879017] Out of memory: Killed process 12547 (v2ray) total-vm:5333904kB, anon-rss:438940kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:980kB oom_score_adj:0
Thu Sep 22 20:40:08 2022 daemon.info v2ray[6938]: V2Ray 4.43.0 (V2Fly, a community-driven edition of V2Ray.) OpenWrt R1 (go1.17.3 linux/amd64)
Thu Sep 22 20:40:08 2022 daemon.info v2ray[6938]: A unified platform for anti-censorship.
Thu Sep 22 20:40:08 2022 daemon.info v2ray[6938]: 2022/09/22 12:40:08 [Info] main/jsonem: Reading config: /var/etc/v2ray/v2ray.main.json
Thu Sep 22 20:42:40 2022 daemon.info vnstatd[8020]: Info: Traffic rate for "eth1" higher than set maximum 1000 Mbit (20s->2673868800, r149688937 t3662236063, 64bit:0), syncing.
Thu Sep 22 20:43:26 2022 user.notice omr-bypass: Starting OMR-ByPass...
[... cut ...]
Thu Sep 22 22:04:16 2022 user.notice post-tracking-post-tracking: Delete default route to 15.X.X.58 via 192.168.125.1 dev eth4
Thu Sep 22 22:04:33 2022 user.notice post-tracking-post-tracking: wan3 (eth3) switched up
Thu Sep 22 22:04:34 2022 user.notice post-tracking-post-tracking: Interface route not yet set, set route ip r add default via 192.168.137.1 dev eth3 metric 5
Thu Sep 22 22:04:35 2022 user.notice post-tracking-post-tracking: wan4 (eth4) switched up
Thu Sep 22 22:04:36 2022 user.notice post-tracking-post-tracking: Interface route not yet set, set route ip r add default via 192.168.125.1 dev eth4 metric 6

Ysurac, 2022-09-22:
Please use template when you open an issue.
This seems to be a VM, you have some "out of memory" so increase memory of the VM.

ccmks, 2022-09-22:
Are you referring to the memory on OMR itself? What is recommended RAM? I am currently assigning it on 1GB RAM for both VPS and OMR

Ysurac, 2022-09-22:
Should work with 1GB, but doesn't seems in your case. V2Ray use more CPU/RAM than Shadowsocks.
You can try 2GB RAM, and check if it's better.

ccmks, 2022-09-22:
How about the VPS? Do you think 1GB RAM should be OK for VPS?

Thank you

Ysurac, 2022-09-24:
Check memory usage on VPS with "htop" for example.
In most case 1GB is enough on both side, but memory usage can be increased in some case (p2p, high speed usage,...)

ccmks, 2022-09-25:
Looks like I have increased the memory on OMR side and keeping VPS  memory on 1GB works fine. I will close this issue now.

Thank you

closed 2022-09-25 by ccmks