Ysurac/openmptcprouter #2567: MPTCP stopped working "You have to install the patched Kernel to use MPTCP" on VPS server

ioogithub, 2022-09-23:
## Expected Behavior

MPTCP stopped working suddenly. All traffic is currently going out only over the default master wan.

## Current Behavior

I did an `omr-test-speed` and monitored the bandwidth graph: `Network->MPTCP->Bandwidth`. I noticed 0 traffic going out though wan1 (wan2 is the default master).

When I did a `Network->MPTCP->Network Support Check` I noticed I did not get a "TCPOptionMPTCPCapable [...] Sender's Key" from either wan1 or wan1.

Next I did an `omr-test-speed` on OMR and  `journalctl -f` on VPS server. I see these messages:
```
Sep 23 02:58:09 vps OMR-Service[21441]: ss-server not detected, restart Shadowsocks
Sep 23 02:58:09 vps systemd[1]: Stopping Shadowsocks-Libev Custom Manager Service for manager...
Sep 23 02:58:09 vps systemd[1]: shadowsocks-libev-manager@manager.service: Succeeded.
Sep 23 02:58:09 vps systemd[1]: Stopped Shadowsocks-Libev Custom Manager Service for manager.
Sep 23 02:58:09 vps systemd[1]: Started Shadowsocks-Libev Custom Manager Service for manager.
Sep 23 02:58:09 vps ss-manager[21443]:  2022-09-23 02:58:09 INFO: using tcp fast open
Sep 23 02:58:09 vps ss-manager[21443]:  2022-09-23 02:58:09 INFO: using tcp no-delay
Sep 23 02:58:09 vps ss-manager[21443]:  2022-09-23 02:58:09 INFO: using MPTCP
Sep 23 02:58:09 vps ss-manager[21443]:  2022-09-23 02:58:09 INFO: running from root user
Sep 23 02:58:09 vps ss-manager[21443]:  2022-09-23 02:58:09 INFO: working directory points to /root/.shadowsocks
Sep 23 02:58:09 vps ss-manager[21443]:  2022-09-23 02:58:09 INFO: try to bind interface: 0.0.0.0, port: 65101
Sep 23 02:58:09 vps ss-manager[21447]:  2022-09-23 02:58:09 INFO: initializing acl...
Sep 23 02:58:09 vps ss-manager[21447]:  2022-09-23 02:58:09 INFO: enable TCP no-delay
Sep 23 02:58:09 vps ss-manager[21447]:  2022-09-23 02:58:09 INFO: enable multipath TCP
Sep 23 02:58:09 vps ss-server[21451]: resolving hostname to IPv6 address first
Sep 23 02:58:09 vps ss-server[21451]: using tcp fast open
Sep 23 02:58:09 vps ss-server[21451]: UDP relay enabled
Sep 23 02:58:09 vps ss-server[21451]: enable TCP no-delay
Sep 23 02:58:09 vps ss-server[21451]: initializing ciphers... aes-256-gcm
Sep 23 02:58:09 vps ss-server[21451]: tcp server listening at 0.0.0.0:65101
Sep 23 02:54:00 vps ss-server[18034]: failed to listen on any address
Sep 23 02:54:00 vps omr-service[18047]: Your device don't support multipath-TCP.
```

This part is repeated many times:
```
Sep 23 02:34:32 vps omr-service[1775]: You have to install the pached kernel to use MPTCP.
Sep 23 02:34:32 vps omr-service[1775]: See http://multipath-tcp.org/ for details
```

Everything is normal with green checkmarks on `System->OpenMPTCProuter->Status` but MPTCP just seemed to stop working.

## Possible Solution

I have no idea about solutions. MPTCP has been running fine all week. Nothing has changed on the VPS server since I originally ran the script. The install script ran and completed without error and I have been using OMR successfully for 5 days until today. The VPS server was not updated or rebooted since the original script ran several days ago.

I rebooted the VPS but same issue.

## Context (Environment)

I am trying to use OpenMPTCProuter normally by aggregating two wan connections. This worked as expected until today. I do not see any changes that could have caused this. I did not update the kernel or install any new software on the VPS instance or the OMR server. I do not see any other unusual behavior on the VPS.

## Specifications

- OpenMPTCProuter version: openmptcprouter v0.59.1-5.4 r0+16594-ce92d
- OpenMPTCProuter VPS version: ? I don't know how to get this. VPS install script ran 5 days ago.
- OpenMPTCProuter VPS provider: digital ocean
- OpenMPTCProuter platform: RPI4b

ioogithub, 2022-09-23:
From researching other issues I found this:

```
root@vps:~# uname -a
Linux vps 5.10.0-11-amd64 #1 SMP Debian 5.10.92-2 (2022-02-28) x86_64 GNU/Linux
```
```
root@vps:~# ls /boot
System.map-5.10.0-11-amd64  config-5.10.0-11-amd64  efi                         initrd.img-5.10.0-18-amd64  vmlinuz-5.10.0-18-amd64
System.map-5.10.0-18-amd64  config-5.10.0-18-amd64  grub                        initrd.img-5.4.207-mptcp    vmlinuz-5.4.207-mptcp
System.map-5.4.207-mptcp    config-5.4.207-mptcp    initrd.img-5.10.0-11-amd64  vmlinuz-5.10.0-11-amd64
```

So for some reason I am not running the mptcp kernel anymore. Yes I did reboot after the initial install and I have been testing OMR successfully for 5 days with this VPS. What can cause the kernel to disappear like this? How can I recover?

Ysurac, 2022-09-23:
You have maybe a script updating your system that is running, no idea...
You can run the install script again.

suyuan168, 2022-09-23:
Maybe your vps automatically updated the kernel to overwrite mptcp.

ioogithub, 2022-09-23:
> Maybe your vps automatically updated the kernel to overwrite mptcp.

Okay so this is apparently what happened. The VPS manages the VM instances and apparently applies updates to the image. An update was applied to my instance with a newer kernel and the policy is that the instances will always boot the latest kernel by default. Since the new kernel was newer than my mptcp kernel it booted into that one.

It is possible to override this behavior set the grub default to the kernel of your choice.

If anyone else is using digitalocean this is the document that will fix the issue: https://docs.digitalocean.com/products/droplets/how-to/kernel/use-non-default/#set-default

I am not sure if other VPS do something similar but hopefully if another use is in this position they can find this issue and be aware of it.

closed 2022-09-23 by ioogithub