Ysurac/openmptcprouter #2570: NanoPi R4S 1GB ram

Silvanskii, 2022-09-24:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

**Describe why you are asking for this support?**

**Additional context**

Hi, I have a Nanopi r4s with 1 gb ram. As we know there are no system images for it now, from the 4gb version they don't work.
How can i help you make support for this device, maybe i should pay you for this device or help the project in some other way?

Ysurac, 2022-09-24:
Duplicate of #2484 
After some research this seems to be only an uboot problem.
Did you try both OpenMPTCProuter 5.4 and 5.15 release ?
If yes, I will try to update uboot.

Silvanskii, 2022-09-24:
> Duplicate of #2484
> After some research this seems to be only an uboot problem.
> Did you try both OpenMPTCProuter 5.4 and 5.15 release ?
> If yes, I will try to update uboot.

I tried all versions as nanopi r4s support appeared. It doesn't turn on with more than one version

Silvanskii, 2022-09-24:
After turning the device on, the boot indicators do not blink, it can not even start booting. Since there is no screen on it I can not see the error log. Many versions of openwrt also do not start, I was able to start only with the version of immortal wrt

Silvanskii, 2022-10-02:
> Duplicate of #2484
> After some research this seems to be only an uboot problem.
> Did you try both OpenMPTCProuter 5.4 and 5.15 release ?
> If yes, I will try to update uboot.

Is there anything I can do to help you make it possible to work on the 1gb version?

Silvanskii, 2022-10-11:
> Duplicate of #2484 After some research this seems to be only an uboot problem. Did you try both OpenMPTCProuter 5.4 and 5.15 release ? If yes, I will try to update uboot.

Is there anything I can do to help you make it possible to work on the 1gb version?

Silvanskii, 2022-10-14:
I got a version for nanopi r4s 1gb ram, a third-party developer helped

closed 2022-10-14 by Silvanskii