Ysurac/openmptcprouter #2572: WhatsApp Can't Make Call Part 2

ccmks, 2022-09-25:
I would like to re-open this issue since the old one was buried already

https://github.com/Ysurac/openmptcprouter/issues/2501

I am using V2RAY and have updated on verson v0.59.1-5.4 and VPS is 1028

Here is the result on Twilio Network Test:

[10:52:19 AM] Test "TURN UDP Connectivity" started...
[10:52:24 AM] Error: Error: Could not establish a UDP connection to Twilio within 5 seconds
[10:52:24 AM] Test "TURN TCP Connectivity" started...
[10:52:26 AM] Successfully established a TCP connection to Twilio in 1621ms
[10:52:26 AM] Test "TURN TCP Connectivity" ended
[10:52:26 AM] Test "TURN TLS Connectivity" started...
[10:52:28 AM] Successfully established a TLS connection to Twilio in 1809ms
[10:52:28 AM] Test "TURN TLS Connectivity" ended
[10:52:28 AM] Test "Bandwidth" started...
[10:52:29 AM] Successfully established a connection to Twilio in 1334ms
[10:53:43 AM] Highest bandwidth: NaN kBits/sec.
[10:53:43 AM] Lowest bandwidth: 8.19 kBits/sec.
[10:53:43 AM] Average bandwidth: 261.46 kBits/sec.
[10:53:43 AM] Supported simultaneous Twilio Client connections:
PCMU (Estimated 100 kBits/sec per connection):
	Average: 2
	Lowest: 0
	Highest: NaN
Opus (Estimated 50 kBits/sec per connection):
	Average: 5
	Lowest: 0
	Highest: NaN
[10:53:43 AM] Test "Bandwidth" ended
[10:53:43 AM] Test "Test call using PCMU" started...
[10:53:43 AM] Initializing Twilio Client. Setting up Twilio.Device using PCMU...
[10:53:44 AM] Twilio Client initialized
[10:53:44 AM] Twilio Client connected to sg1
[10:53:44 AM] Requesting microphone access...
[10:53:44 AM] Call connected with CallSid CAd939aa0b2a9c893e62a495ae8ff5e89d
[10:53:53 AM] Call was hung up
[10:53:53 AM] Twilio Client disconnected. Overall quality not available - not supported by browser.
[10:53:54 AM] Error: Error: Could not establish an ICE connection within 10 seconds.
[10:53:54 AM] Test "Test call using Opus" started...
[10:53:54 AM] Initializing Twilio Client. Setting up Twilio.Device using Opus...
[10:53:55 AM] Twilio Client initialized
[10:53:55 AM] Twilio Client connected to sg1
[10:53:55 AM] Requesting microphone access...
[10:53:55 AM] Call connected with CallSid CAc6afb517f22b518446ba56822c048025
[10:54:04 AM] Call was hung up
[10:54:04 AM] Twilio Client disconnected. Overall quality not available - not supported by browser.
[10:54:05 AM] Error: Error: Could not establish an ICE connection within 10 seconds.
[10:54:05 AM] Test "Insights Connectivity" started...
[10:54:06 AM] Successfully established a connection to Twilio Insights endpoint in 989ms
[10:54:06 AM] Test "Insights Connectivity" ended
[10:54:06 AM] Test "Signaling Connectivity" started...
[10:54:06 AM] creating room...
[10:54:10 AM] Successfully joined a Room
[10:54:10 AM] Room SID: RMbfcfd6f7562613aab3da16eb97407fbd
[10:54:12 AM] Successfully established a connection to Twilio signaling endpoint
[10:54:12 AM] Test "Signaling Connectivity" ended
[10:54:12 AM] Test "Test Group Room with TURN" started...
[10:54:12 AM] Attempting to connect to a Room
[10:54:17 AM] Successfully joined a Room
[10:54:17 AM] Room SID:RM03476a63dffbd26fbe26a4810dcd67c8
[10:54:22 AM] BytesReceived (Audio): 12078
[10:54:22 AM] BytesReceived (Video): 16095
[10:54:22 AM] Group Room completed successfully
[10:54:22 AM] Test "Test Group Room with TURN" ended
[10:54:22 AM] Test "Test Group Room without TURN" started...
[10:54:22 AM] Attempting to connect to a Room
[10:54:26 AM] Successfully joined a Room
[10:54:26 AM] Room SID:RM49934dbe8515d681f00833310dc5fb2a
[10:54:31 AM] BytesReceived (Audio): 12919
[10:54:31 AM] BytesReceived (Video): 17307
[10:54:31 AM] Group Room completed successfully
[10:54:31 AM] Test "Test Group Room without TURN" ended
[10:54:31 AM] Done.

Any idea why it is no longer working?

Thank you

Kalimeiro, 2022-09-25:
- download the latest snapshot (0.59.2alpha) and disable v2ray UDP in "advanced settings" of "system" -> "openmptcprouter" ;
- wait 30 seconds then restart the router.

ccmks, 2022-09-25:
V2RAY for streaming and video call is great! Such as Google Meet. Disable UDP in V2ray will lose this benefit?

Kalimeiro, 2022-09-25:
> Disable UDP in V2ray will lose this benefit?

No.

ccmks, 2022-09-25:
So what's the difference disabling UDP in V2RAY, in v0.59.1 vs v0.59.2 alpha?

Kalimeiro, 2022-09-25:
> So what's the difference disabling UDP in V2RAY, in v0.59.1 vs v0.59.2 alpha?

In 0.59.1 you can't disable V2RAY UDP, it's not working.

in 0.59.2alpha this bug is solved.

ccmks, 2022-09-25:
If you disable UDP in V2RAY, then the UDP traffic won't be using MPTCP anymore, since WhatsApp using UDP traffic to do video and voice call

Kalimeiro, 2022-09-25:
> If you disable UDP in V2RAY, then the UDP traffic won't be using MPTCP anymore, since WhatsApp using UDP traffic to do video and voice call

Wrong. it use the VPN with Glorytun for UDP/ICMP/all different than TCP, and MPTCP is working.

You really need spend time to test the solution, it takes 15 minutes of your time to apply the snapshot upgrade, disable V2RAY UDP, restart the router, test twilio.

Bye.

ccmks, 2022-09-25:
Try the test again tomorrow, It happened to me last time. After make changes, it was working OK, until several hours or even next day, it didn't work again.  If you read my previous post, I have stated it was working then proceeded closing the issue.

However, it stop working again after several days (that's when I started to notice it)

I tried to use VPN for UDP like glorytun, but it is not as smooth as using V2RAY  when transition from WAN1 to WAN2 disconnection even when set to MPTCP scheduler to REDUNDANT.

That's why V2RAY UDP is like God's gift for us.

@Ysurac what do you think?

Thank you for making this project happen as this is the answer for our internet in our country where stable and reliable internet is not affordable and not much covered area.

Thank you

molesza, 2022-12-12:
I also have problem with Whatsapp calls not connecting. Can someone point me in the right direction to get 0.59.2alpha?

molesza, 2022-12-20:
I have upgraded to [0.59.1-5.4 r0+16594 and my whatsapp calls are now working properly. Did not have to change anything. No OMR bypass even. Great stuff!

closed 2023-03-26 by github-actions[bot]