Ysurac/openmptcprouter #2575: Is Banana Pi M2 Plus supported?

mahdiyari, 2022-09-25:
This board has 1Gbit ethernet and is very cheap compared to raspberry pi (5-6 times cheaper). Can I buy and use it as my OMR?
I didn't see any mentions about this board on here so hence the issue.
I never had a router nor installed one. Complete newbie here.

I did find this on OpenWrt https://openwrt.org/toh/sinovoip/banana_pi_m2_plus

I really appreciate what you do here. Great job.

Ysurac, 2022-09-26:
I only support hardware that I own.
I can buy them but I'm not sure CPU of these boards have speed of a RPI4, I would need to find a hardware tests.
Actually x86_64 based board can be cheaper than a RPI4 (and far better).

mahdiyari, 2022-09-26:
I can buy and test the BPI-M2+
Can I install OMR myself on them?

mahdiyari, 2022-09-27:
Excuse my ignorance. I didn't understand what you meant by x86_64 boards at first. I will look for those. If I understand correctly the non-arm cpus like intel etc.

mahdiyari, 2022-09-27:
Will this work? https://support.hp.com/us-en/document/c04356040
Or I need to AIM for a minimum CPU speed/core

closed 2022-09-28 by mahdiyari