Ysurac/openmptcprouter #2577: Raspberrypi filesystem only read after powerloss, can't modify anything

simonemessina92, 2022-09-26:
After raspberrypi loose power sometimes this happen, and I can't solve in anyway, have to flash again the sd card with openmptcp .. Is any way to solve this problem ? 
[image]

Ysurac, 2022-09-26:
Please use template when you open an issue.
Filesystem is put as read only after error on FS. This can be due to hard reboot or bad SD card.
You need to reinstall RPI, and if it's not already the case check "Check filesystems before mount" in System->Mount Points.

simonemessina92, 2022-09-26:
> Please use template when you open an issue. Filesystem is put as read only after error on FS. This can be due to hard reboot or bad SD card. You need to reinstall RPI, and if it's not already the case check "Check filesystems before mount" in System->Mount Points.

Sorry Yusrac, I'm running rpi 4 4gb with omr latest 0.59.1-5.4   This error happen too many times I've tried diffrent sd cards not always with the same, so I wonder if there's something like a bug or other things. The check filesystem before mount is checked since the first installation.

Ysurac, 2022-09-28:
Use an inverter if it's only after power outage.
Make sure you don't have any USB devices, this can also use too much power for the RPI.
You can also try to flash the RPI4, this can sometimes solve RPI issues: https://github.com/Ysurac/openmptcprouter/wiki/Update-RPI4-firmware

Kalimeiro, 2022-09-28:
> Sorry Yusrac, I'm running rpi 4 4gb with omr latest 0.59.1-5.4 This error happen too many times I've tried diffrent sd cards not always with the same, so I wonder if there's something like a bug or other things. The check filesystem before mount is checked since the first installation.

I'm agree with this comment part.

It's not always a SD CARD faulty, it's an OMR problem only on RPI4, for now it's working great with 0.59.1, but sometimes, the FS is ready only when we have short electrical.

I put the SD CARD on my laptop, execute manually an e2fsck to correct bad inode and fix other issue, then i get file in /etc/config with 0 bytes. So clearly when this problem happen : OMR is writting to the SDCARD, short electrical damage the FS, and after reboot the check FS can't fix automatically the issue.

Ysurac, 2022-09-29:
You may try squashfs image, it write less often on SD CARD, but not sure this really change something for power outage.

Silvanskii, 2022-10-11:
I solved this problem by doing a full format of the sd card and then installing the squashfs version. Not a quick format but a full format

iamdivyanshverma, 2022-12-23:
instead of full format , you can try 
-> e2fsck -y /dev/mmcblk0p2
-> reboot

closed 2023-03-29 by github-actions[bot]