Ysurac/openmptcprouter #2579: depends libuv1

Danymp1974, 2022-09-26:
during installation on debian 11 it crashes with the following dependency error

bind9-libs : Depends: libuv1 (>= 1.40.0) but 1.34.2-1 is to be installed
E: Unable to correct problems, you heve held broken packages

best regards

Ysurac, 2022-09-26:
Please use the template when you open an issue.
What is the VPS provider ? It's a fresh install or VPS is also used for something else ?
I would need more log.

Danymp1974, 2022-09-26:
the installation is on a vsphere 6.7 virtual machine, clean Debian 11 operating system used for this application only

redkite127, 2022-09-26:
I have the same problem on a freshly installed OVH VPS.
If I download and install `libuv1` from https://packages.debian.org/bullseye/libuv1 I can successfully install `bind9-libs` with apt.
But I guess that after, there is a dist-upgrade downgrading and removing what's installed manually :-/

```
# apt dist-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  libfstrm0 libicu67 liblmdb0 libmaxminddb0 libprotobuf-c1 libxml2
Use 'sudo apt autoremove' to remove them.
The following packages will be DOWNGRADED:
  python3-wsproto
0 upgraded, 0 newly installed, 1 downgraded, 0 to remove and 0 not upgraded.
```

Kalimeiro, 2022-09-26:
Same problem with same package. Working with Debian 10.

Danymp1974, 2022-09-27:
I confirm with the Debian distribution 10 all ok

Ysurac, 2022-09-27:
Fixed.

closed 2022-09-27 by Danymp1974

simon2cyrene, 2022-10-13:
Failed to install pi-hole on Bullseye. I think it's related to this dependency.
The script requires the installation of dnsutils which fails because bind9-libs depends on libuv1 > 1.40.

> `The following packages have unmet dependencies:
>  bind9-libs : Depends: libuv1 (>= 1.40.0) but 1.34.2-1 is to be installed
> E: Unable to correct problems, you have held broken packages.
> `

Is it possible to modify `/etc/apt/preferences.d/openmptcprouter.pref` ?

Ysurac, 2022-10-25:
Should be fixed now. You need to run the script again and the .pref is modified for Debian 11.

simon2cyrene, 2022-10-28:
> Should be fixed now. You need to run the script again and the .pref is modified for Debian 11.

Thanks. Everything is fine on a new VPS installation.

joshenders, 2022-11-01:
Still seeing this issue on a fresh install of Debian 11 and HEAD of the master branch of https://github.com/Ysurac/openmptcprouter-vps/blob/develop/debian9-x86_64.sh

@Ysurac Can you link to the commit that supposedly fixed this issue?