Ysurac/openmptcprouter #2580: Aggregated speed slower than single values

realgpp, 2022-09-26:
## Expected Behavior

I'm trying to aggregate two connections (100M ADSL and ~70M mobile router) with EdgeRouter X but I get really slow download values.

## Current Behavior

I've configured 100M ADSL on eth1 as Master for "Multipath TCP" and mobile on eth2 as Enabled for "Multipath TCP". I've left any other configurations untouched included the VPS (which is on Aruba.it).
As you can see below the aggregated speed is even lower than the mobile connection itself (5,6k from test).
I don't known if correlated or not but when I plug the cable of eth0 I cannot surf the internet
In attachment the content of tab openmptcprouter > Show all settings if it can help.

Do you have any suggestion?

root@OpenMPTCProuter:~#  omr-speedtest  eth1
Download server list...
Select best server...
..........
Done: url: http://speedtest.cheapnet.it:8080/speedtest/upload.php - host: speedtest.cheapnet.it:8080 - ping: 21
Download test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 92.4M  100 92.4M    0     0  11.5M      0  0:00:07  0:00:07 --:--:-- 11.8M

root@OpenMPTCProuter:~#  omr-speedtest  eth2
Download server list...
Select best server...
..........
Done: url: http://test.intred.it:8080/speedtest/upload.php - host: test.intred.it:8080 - ping: 204
Download test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 92.4M  100 92.4M    0     0  6002k      0  0:00:15  0:00:15 --:--:-- 5627k

root@OpenMPTCProuter:~#  omr-speedtest
Download server list...
Select best server...
..........
Done: url: http://test.intred.it:8080/speedtest/upload.php - host: test.intred.it:8080 - ping: 93
Download test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 92.4M  100 92.4M    0     0  3046k      0  0:00:31  0:00:31 --:--:-- 3068k

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version: 0.1028, Debian 10, kernel 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Aruba.it
  - OpenMPTCProuter platform: EdgeRouter X 

[image]
[omr-settings.txt](https://github.com/Ysurac/openmptcprouter/files/9648513/omr-settings.txt)

Ysurac, 2022-09-26:
Use ```omr-test-speed``` instead of ```omr-speedtest``` to test connections speeds.
Check also load average during speed tests.
As Edgerouter have a slow CPU, you can try to disable encryption in wizard, "advanced settings" checkbox.

realgpp, 2022-09-27:
Even with encryption disabled I don't see any break through improvement.
It seems it didn't have any kind of impact, which it seems very strange imho. Are there any other configurations I can tweak or some other checks I can do?

#omr-test-speed eth1
Select best test server...
host: scaleway.testdebit.info - ping: 28
host: bordeaux.testdebit.info - ping: 51
host: aix-marseille.testdebit.info - ping: 44
host: lyon.testdebit.info - ping: 41
host: lille.testdebit.info - ping: 32
host: paris.testdebit.info - ping: 35
host: appliwave.testdebit.info - ping: 28
host: speedtest.frankfurt.linode.com - ping: 20
host: speedtest.tokyo2.linode.com - ping: 261
host: speedtest.singapore.linode.com - ping: 249
host: speedtest.newark.linode.com - ping: 106
host: speedtest.atlanta.linode.com - ping: 147
host: speedtest.dallas.linode.com - ping: 130
host: speedtest.fremont.linode.com - ping:
host: speed.hetzner.de - ping: 23
host: ipv4.bouygues.testdebit.info - ping: 36
host: par.download.datapacket.com - ping: 30
host: nyc.download.datapacket.com - ping: 98
host: ams.download.datapacket.com - ping: 29
host: fra.download.datapacket.com - ping: 21
host: lon.download.datapacket.com - ping: 33
host: mad.download.datapacket.com - ping: 47
host: prg.download.datapacket.com - ping: 25
host: sto.download.datapacket.com - ping: 40
host: vie.download.datapacket.com - ping: 30
host: war.download.datapacket.com - ping: 39
host: atl.download.datapacket.com - ping: 137
host: chi.download.datapacket.com - ping: 137
host: lax.download.datapacket.com - ping: 174
host: mia.download.datapacket.com - ping: 130
host: nyc.download.datapacket.com - ping: 97
host: speedtest.milkywan.fr - ping: 27
Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024M    0 1024M    0     0  11.8M      0 --:--:--  0:01:26 --:--:-- 11.9M
(interrupted)

# omr-test-speed eth2
Select best test server...
host: scaleway.testdebit.info - ping: 46
host: bordeaux.testdebit.info - ping: 51
host: aix-marseille.testdebit.info - ping: 38
host: lyon.testdebit.info - ping: 56
host: lille.testdebit.info - ping: 48
host: paris.testdebit.info - ping: 39
host: appliwave.testdebit.info - ping: 43
host: speedtest.frankfurt.linode.com - ping: 32
host: speedtest.tokyo2.linode.com - ping: 269
host: speedtest.singapore.linode.com - ping: 177
host: speedtest.newark.linode.com - ping: 102
host: speedtest.atlanta.linode.com - ping: 136
host: speedtest.dallas.linode.com - ping: 159
host: speedtest.fremont.linode.com - ping: 179
host: speed.hetzner.de - ping: 38
host: ipv4.bouygues.testdebit.info - ping: 58
host: par.download.datapacket.com - ping: 39
host: nyc.download.datapacket.com - ping: 104
host: ams.download.datapacket.com - ping: 38
host: fra.download.datapacket.com - ping: 26
host: lon.download.datapacket.com - ping: 45
host: mad.download.datapacket.com - ping: 51
host: prg.download.datapacket.com - ping: 35
host: sto.download.datapacket.com - ping: 69
host: vie.download.datapacket.com - ping: 37
host: war.download.datapacket.com - ping: 52
host: atl.download.datapacket.com - ping: 141
host: chi.download.datapacket.com - ping: 128
host: lax.download.datapacket.com - ping: 185
host: mia.download.datapacket.com - ping: 150
host: nyc.download.datapacket.com - ping: 121
host: speedtest.milkywan.fr - ping: 40
Best server is http://fra.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 14 9536M   14 1341M    0     0  6938k      0  0:23:27  0:03:17  0:20:10 6887k

# omr-test-speed
Select best test server...
host: scaleway.testdebit.info - ping: 63
host: bordeaux.testdebit.info - ping: 49
host: aix-marseille.testdebit.info - ping: 47
host: lyon.testdebit.info - ping: 41
host: lille.testdebit.info - ping: 46
host: paris.testdebit.info - ping: 54
host: appliwave.testdebit.info - ping: 50
host: speedtest.frankfurt.linode.com - ping: 46
host: speedtest.tokyo2.linode.com - ping: 251
host: speedtest.singapore.linode.com - ping: 185
host: speedtest.newark.linode.com - ping: 135
host: speedtest.atlanta.linode.com - ping: 146
host: speedtest.dallas.linode.com - ping: 171
host: speedtest.fremont.linode.com - ping: 189
host: speed.hetzner.de - ping: 43
host: ipv4.bouygues.testdebit.info - ping: 59
host: par.download.datapacket.com - ping: 44
host: nyc.download.datapacket.com - ping: 120
host: ams.download.datapacket.com - ping: 43
host: fra.download.datapacket.com - ping: 39
host: lon.download.datapacket.com - ping: 49
host: mad.download.datapacket.com - ping: 56
host: prg.download.datapacket.com - ping: 42
host: sto.download.datapacket.com - ping: 70
host: vie.download.datapacket.com - ping: 41
host: war.download.datapacket.com - ping: 64
host: atl.download.datapacket.com - ping: 134
host: chi.download.datapacket.com - ping: 140
host: lax.download.datapacket.com - ping: 177
host: mia.download.datapacket.com - ping: 147
host: nyc.download.datapacket.com - ping: 123
host: speedtest.milkywan.fr - ping: 40
Best server is http://fra.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  6 9536M    6  628M    0     0  3781k      0  0:43:02  0:02:50  0:40:12 4332k
(interrupted)

ccmks, 2022-09-27:
Can you SSH to your OMR, and run htop command to check the CPU usage, try to download test while still checking your CPU, if it is reaching 100%, that means, the CPU is bottleneck, which prevent you to go any faster.

I am suspecting this device can only go up to 60 Mbps max

realgpp, 2022-09-28:
I can confirm CPU are at 90-100 when a speed test is launched wihch is a bummer because a buyed the device only for this purpose. Maybe it would be helpful for people after me who approch this new world to add a warning next to the image download which clearly states "device can achive aggregated download speed of 60Mbps" or similar

From [here](https://www.openmptcprouter.com/download): Ubiquiti Edgerouter X (alpha support with issues, max aggregated download speed of 60Mbps)

Not complaining here but imho transparency is always the right way than people will make his considerations :)

ccmks, 2022-09-28:
Edgerrouter X is indeed has slow CPU, therefore it is not quite stable yet and ysurac mark it as alpha. 

I was about to buy that router too, but given consideration that was still in alpha state, I decided not to buy that.

realgpp, 2022-10-06:
Hi, I've changed hardware and buyed a Mini PC with 8 GB DDR4, Celeron J4125 with support to AES-NI, 2 Gigabit Eth.
In terms of configuration I'm currently using one eth for 100MB line and usb0 for tethering.
I've tried different configuration:

- Encryption: None, AES-256-GCM, AES-256-CFB, chacha20
- Enable SQM: True/False
- Enable SQM autorate: True for usb0
- Congestion Control: bbr, cubic

Sadly I'm not able to go over 25Mbps.  
I've monitored the CPU during a test and it's under 25% on average with very fast spike at 60%.
What else can I try?

**omr-test-speed**
[..]
Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024M    0 1024M    0     0  19.8M      0 --:--:--  0:00:51 --:--:-- 30.8M

**omr-test-speed eth1**
[..]
Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024M    0 1024M    0     0  11.1M      0 --:--:--  0:01:32 --:--:-- 11.1M

**omr-test-speed usb0**
[..]
Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  566M    0  566M    0     0  14.3M      0 --:--:--  0:00:39 --:--:-- 17.2M^C

Ysurac, 2022-10-07:
The results of omr-test-speed are in Mo/s, so it's not 30.8Mb/s but 30.8*8=246Mb/s here.

closed 2023-01-10 by github-actions[bot]

Ideonium, 2023-03-04:
I can confirm a similar issue. 
I am running openmptcprouter v0.59.1-5.4 r0+16594 with 3 LTE WAN gateways.
Speedtest on each WAN shows me:
WAN1 ~ 900kbit
WAN2 ~ 800kbit
WAN3 ~ 760kbit
But when combined I get only 650kbit.
When I unplug the cable from one of the Gateways and retest with only 2 active, I get better speed than with all 3!
And even better, When I unplug the cable from the second WAN and leave only one (the main WAN1) I get more than with 2!

**So in my case two gateways are slower than one, and three are even worse...**

OpenMPTCProuter is running on dual core intel Celeron laptop.  1GB RAM (only 180MB used)
top shows CPU usage at around 8% (load average during test)
VPS server provider =OVH

Ysurac, 2023-03-04:
@Ideonium open your own issue using the template, this issue is closed.
And paste omr-test-speed results and status page screenshot.