Ysurac/openmptcprouter #2582: OpenVPN Bonding is not working

ccmks, 2022-09-27:
## Expected Behavior
OpenVPN Bonding is working

## Current Behavior
It is not working when selecting OpenVPN-Bonding, it keep saying VPN tunnel Down on status page

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.207
  - OpenMPTCProuter platform: x86_64

I will attach the log below

ccmks, 2022-09-27:
Tue Sep 27 16:10:23 2022 user.notice ucitrack: Setting up /etc/config/firewall reload dependency on /etc/config/sqm-autorate
Tue Sep 27 16:10:23 2022 user.notice ucitrack: Setting up /etc/config/dhcp reload dependency on /etc/config/odhcpd
Tue Sep 27 16:10:23 2022 user.notice ucitrack: Setting up non-init /etc/config/fstab reload handler: /sbin/block mount
Tue Sep 27 16:10:23 2022 user.notice ucitrack: Setting up /etc/config/system reload trigger for non-procd /etc/init.d/led
Tue Sep 27 16:10:23 2022 daemon.notice openvpn(omr_bonding_wan1)[18301]: TUN/TAP device bondwan1 opened
Tue Sep 27 16:10:23 2022 daemon.notice openvpn(omr_bonding_wan1)[18301]: /usr/libexec/openvpn-hotplug up omr_bonding_wan1 bondwan1 1500 1656   init
Tue Sep 27 16:10:23 2022 user.notice ucitrack: Setting up /etc/config/system reload dependency on /etc/config/luci_statistics
Tue Sep 27 16:10:23 2022 user.notice ucitrack: Setting up /etc/config/system reload dependency on /etc/config/dhcp
Tue Sep 27 16:10:23 2022 daemon.notice openvpn(omr_bonding_wan1)[18301]: Initialization Sequence Completed
Tue Sep 27 16:10:23 2022 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/glorytun-udp
Tue Sep 27 16:10:23 2022 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/openmptcprouter-vps
Tue Sep 27 16:10:23 2022 user.notice ucitrack: Setting up /etc/config/glorytun-udp reload dependency on /etc/config/openmptcprouter-vps
Tue Sep 27 16:10:23 2022 daemon.notice openvpn(omr_bonding_wan2)[18519]: TUN/TAP device bondwan2 opened
Tue Sep 27 16:10:23 2022 daemon.notice openvpn(omr_bonding_wan2)[18519]: /usr/libexec/openvpn-hotplug up omr_bonding_wan2 bondwan2 1500 1656   init
Tue Sep 27 16:10:23 2022 daemon.notice openvpn(omr_bonding_wan2)[18519]: Initialization Sequence Completed
Tue Sep 27 16:10:23 2022 user.notice ucitrack: Setting up /etc/config/shadowsocks-libev reload dependency on /etc/config/omr-tracker
Tue Sep 27 16:10:23 2022 user.notice ucitrack: Setting up /etc/config/mlvpn reload trigger for non-procd /etc/init.d/mlvpn
Tue Sep 27 16:10:23 2022 daemon.err openvpn(omr_bonding_wan3)[18770]: write UDP: Operation not permitted (code=1)
Tue Sep 27 16:10:23 2022 daemon.err modprobe: failed to find a module named sch_fq_codel
Tue Sep 27 16:10:23 2022 daemon.warn [8856]: <warn>  [SQM_IFB_9344e] invalid sysfs path read for net/SQM_IFB_9344e
Tue Sep 27 16:10:23 2022 daemon.warn [8856]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_9344e not found
Tue Sep 27 16:10:23 2022 daemon.notice openvpn(omr_bonding_wan4)[19040]: TUN/TAP device bondwan4 opened
Tue Sep 27 16:10:23 2022 daemon.notice openvpn(omr_bonding_wan4)[19040]: /usr/libexec/openvpn-hotplug up omr_bonding_wan4 bondwan4 1500 1656   init
Tue Sep 27 16:10:23 2022 daemon.notice openvpn(omr_bonding_wan4)[19040]: Initialization Sequence Completed
Tue Sep 27 16:10:23 2022 daemon.err modprobe: failed to find a module named sch_fq_codel
Tue Sep 27 16:10:24 2022 daemon.warn [8856]: <warn>  [SQM_IFB_bfad6] invalid sysfs path read for net/SQM_IFB_bfad6
Tue Sep 27 16:10:24 2022 daemon.warn [8856]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_bfad6 not found
Tue Sep 27 16:10:24 2022 user.notice firewall: Reloading firewall due to ifup of wan4 (eth4)
Tue Sep 27 16:10:24 2022 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Tue Sep 27 16:10:25 2022 user.notice mptcp: Reloading mptcp config due to ifup of wan4 (eth4)
Tue Sep 27 16:10:25 2022 daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Tue Sep 27 16:10:25 2022 daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Tue Sep 27 16:10:25 2022 daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Tue Sep 27 16:10:25 2022 daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Tue Sep 27 16:10:25 2022 daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Tue Sep 27 16:10:26 2022 user.notice firewall: Reloading firewall due to ifup of wan5 (eth5)
Tue Sep 27 16:10:26 2022 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Tue Sep 27 16:10:26 2022 user.notice omr-tracker: Launching...
Tue Sep 27 16:10:27 2022 user.notice mptcp: Reloading mptcp config due to ifup of wan5 (eth5)
Tue Sep 27 16:10:28 2022 user.notice firewall: Reloading firewall due to ifup of wan6 (eth6)
Tue Sep 27 16:10:28 2022 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Tue Sep 27 16:10:28 2022 user.notice mptcp: Reloading mptcp config due to ifup of wan6 (eth6)
Tue Sep 27 16:10:29 2022 user.notice firewall: Reloading firewall due to ifup of wan7 (eth7)
Tue Sep 27 16:10:29 2022 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Tue Sep 27 16:10:30 2022 user.notice mptcp: Reloading mptcp config due to ifup of wan7 (eth7)
Tue Sep 27 16:10:31 2022 user.notice firewall: Reloading firewall due to ifup of wan8 (eth8)
Tue Sep 27 16:10:31 2022 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Tue Sep 27 16:10:31 2022 user.notice mptcp: Reloading mptcp config due to ifup of wan8 (eth8)
Tue Sep 27 16:10:36 2022 user.notice omr-tracker: Launched
Tue Sep 27 16:10:36 2022 user.notice SQM-autorate: Launch on wan1
[... cut ...]
Tue Sep 27 20:34:05 2022 daemon.err openvpn(omr_bonding_wan4)[19040]: TLS Error: TLS handshake failed
Tue Sep 27 20:34:10 2022 daemon.err /usr/bin/ss-redir[27986]: server recv: Host is unreachable
Tue Sep 27 20:35:20 2022 daemon.err openvpn(omr_bonding_wan4)[19040]: TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)
Tue Sep 27 20:35:20 2022 daemon.err openvpn(omr_bonding_wan4)[19040]: TLS Error: TLS handshake failed
Tue Sep 27 20:35:31 2022 daemon.err /usr/bin/ss-redir[27986]: send: Connection reset by peer

Ysurac, 2022-09-28:
This will be removed in next release as it's never working correctly even when key handshake is working.