Ysurac/openmptcprouter #2586: Hardware setup help

realgpp, 2022-09-29:
In case of pi4 or any other x86 board with only one Ethernet port how is it possible to use an Ethernet AP to make other devices connect to leverage aggregated speed by OMR?
Is it possible to use a USB-RJ45 adapter to link AP and OMR? 
Or is it mandatory to have a switch (with power supply)? In the latter case, is it possible to configure OMR to use one external connection (DSL or 4g) for each port of the switch?

Is it true it's better to have eth ports with Intel chipset?

Ysurac, 2022-09-29:
Better not to use any USB-RJ45 adapters.
You need an ethernet switch (they always have power supply or they use PoE), and with a RPI you can aggregate up to 8 connection. For an AP, it only need to use OpenMPTCProuter IP as gateway (it's the case when it use DHCP).

Not really true for Intel chipset, for example I22x have some problems for now with OpenMPTCProuter.

ahayes, 2022-09-29:
@realgpp I just recently set up OMR. I had the same concerns about using a Raspberry Pi 4 with single ethernet adapter because my aggregated bandwidth would theoretically be higher than a single gigabit port could handle.

In my case, I bought a small Intel box with four separate Intel I226-V adapter ports and have had to use a test build of OMR to have them recognized at all, and I and others have also had to find some workarounds due to DHCP issues with the Intel I22x adapters. It seems like **maybe** the issues have been resolved upstream in more recent Linux kernels but it may take a while for the fix to appear in OpenWRT and OMR.

But since working around the DHCP issue, things have been working great and my combined DSL and cable internet connection is solid and all appears from a single fixed VPS IP so even Plex is happy! @Ysurac is my new hero.

ioogithub, 2022-09-30:
I have a similar box with 4x I226-V NICs. The issue is here: https://github.com/Ysurac/openmptcprouter/issues/2584 however since I checked the "force link" setting it has been stable ever since. This "force link" settings is default for static but not for DHCP so being aware of this check seems to be the solution.

This type of device simplifies things considerably without needing a switch or VLANs.

I should mention that after purchasing an x86 box with 4 dedicated NICs has not fixed the bandwidth issues I was seeing with my rpi USB adpaters. My aggregate bandwidth is still much slower than the slowest connection alone and the bandwidth degrades substantially over time until I manually reset the interfaces.

I am still evaluating this hardware purchase but so far it hasn't been a success with OMR.

iambenmitchell, 2022-09-30:
I use a raspberry pi 4 and I get about 130mbps down / 30mbps up. (2x 80/20 VDSL connections).

I might make a YouTube tutorial on how to setup the Vlans and stuff for this configuration if anyone is interested.

vinceducat, 2022-10-01:
@ahayes  i m interrest in witch type of small pc box do you use do you have brand and model ?

ahayes, 2022-10-09:
@vinceducat Sorry for the delay. The vendor I picked was CWWK (changwang.com) on [Ali Express](https://m.aliexpress.com/item/1005004641076475.html). They were recommended by someone else in the comments section of an Serve The Home video because they provide [firmware updates](https://www.changwang.com/down/). They were also very quick to respond to my questions in the AliExpress chat system. They even asked me which colour I wanted as soon as I ordered. Basic postal shipping was fast enough. Note that as we've been finding in this and other issues, there are currently issues with the Intel I225-V and I226-V adapters in these systems. It looks like more recent linux kernels might have a driver that works well but if your goal is OMR then just be aware of the issue above and related issues.

closed 2023-01-13 by github-actions[bot]