Ysurac/openmptcprouter #2592: Banana Pi R3 support ?

iDustbin, 2022-10-04:
Hi, will you create an upgrade to support banana pi r3 ?

Ysurac, 2022-10-07:
I only officially support device that I own.
I will try to support it if SinoVoip send me one ;)

Should be possible using by merging changes in https://github.com/BPI-SINOVOIP/BPI-R3-OPENWRT-V21.02.3

Ysurac, 2022-10-07:
Support is already in OpenWRT https://github.com/openwrt/openwrt/commit/a96382c1bb204698cd43e82193877c10e4b63027
So should work if you compile OpenMPTCProuter 5.15 based release.

polskafan, 2023-01-09:
I'm currently trying to build 5.15 for the Banana Pi R3. Looks like, there isn't much missing, but unfortunately the build fails.

This is what I tried (on branch develop, as master doesn't have 5.15 yet):
```
git clone https://github.com/Ysurac/openmptcprouter.git
cd openmptcprouter
OMR_TARGET="bpi-r3" OMR_FEED_SRC="master" OMR_KERNEL="5.15" ./build.sh
cd bpi-r3/5.15/source
make menuconfig 
```

Then select Mediatek ARM and Filogic 830 MT7986A in menuconfig and ```make -j1 V=s```

That fails when trying to build shortcut-fe:
```
make -f ./scripts/Makefile.build obj=/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe \
single-build= \
need-builtin=1 need-modorder=1
  aarch64-openwrt-linux-musl-gcc -Wp,-MMD,/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe/.sfe_ipv4.o.d -nostdinc -isystem /home/vbox/openmptcprouter/bpi-r3/5.15/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/include -I./arch/arm64/include -I./arch/arm64/include/generated  -I./include -I./arch/arm64/include/uapi -I./arch/arm64/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -mlittle-endian -DKASAN_SHADOW_SCALE_SHIFT= -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu11 -mgeneral-regs-only -DCONFIG_CC_HAS_K_CONSTRAINT=1 -Wno-psabi -mabi=lp64 -fno-asynchronous-unwind-tables -fno-unwind-tables -mbranch-protection=none -Wa,-march=armv8.5-a -DARM64_ASM_ARCH='"armv8.5-a"' -DKASAN_SHADOW_SCALE_SHIFT= -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=1024 -fstack-protector -Wimplicit-fallthrough=5 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-alloc-size-larger-than -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -fmacro-prefix-map=/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves -mstack-protector-guard=sysreg -mstack-protector-guard-reg=sp_el0 -mstack-protector-guard-offset=1040 -DSFE_SUPPORT_IPV6  -DMODULE  -DKBUILD_BASENAME='"sfe_ipv4"' -DKBUILD_MODNAME='"shortcut_fe"' -D__KBUILD_MODNAME=kmod_shortcut_fe -c -o /home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe/sfe_ipv4.o /home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe/sfe_ipv4.c
/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_udp':
/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe/sfe_ipv4.c:1478:12: error: 'struct sk_buff' has no member named 'fast_forwarded'
 1478 |         skb->fast_forwarded = 1;
      |            ^~
/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_tcp':
/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe/sfe_ipv4.c:2052:12: error: 'struct sk_buff' has no member named 'fast_forwarded'
 2052 |         skb->fast_forwarded = 1;
      |            ^~
/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_periodic_sync':
/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe/sfe_ipv4.c:2868:53: error: 'struct timer_list' has no member named 'cust_data'
 2868 |         struct sfe_ipv4 *si = (struct sfe_ipv4 *)arg->cust_data;
      |                                                     ^~
/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_init':
/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe/sfe_ipv4.c:3552:18: error: 'struct timer_list' has no member named 'cust_data'
 3552 |         si->timer.cust_data = (unsigned long)si;
      |                  ^
make[5]: *** [scripts/Makefile.build:289: /home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe/sfe_ipv4.o] Error 1
make[4]: *** [Makefile:1903: /home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/shortcut-fe] Error 2
[... cut ...]
make[2]: *** [package/Makefile:116: package/feeds/openmptcprouter/ndpi-netfilter2/compile] Error 1
make[2]: Leaving directory '/home/vbox/openmptcprouter/bpi-r3/5.15/source'
make[1]: *** [package/Makefile:110: /home/vbox/openmptcprouter/bpi-r3/5.15/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/vbox/openmptcprouter/bpi-r3/5.15/source'
make: *** [/home/vbox/openmptcprouter/bpi-r3/5.15/source/include/toplevel.mk:231: world] Error 2
```

Any ideas? Did I screw up the toolchain with my patches?

polskafan, 2023-01-10:
So I fixed that error with a patch for the Makefile that includes the "-nostartfiles" flag when building the shared object for ndpi-netfilter2/lpt.

After that the build got stuck on missing dependency errors for the firewall package. Took me a while to realize that I was trying to build the openmptcprouter master feed instead of the develop feed for kernel 5.15. So back to square one. At least I am slowly getting familiar with the build system.

polskafan, 2023-01-10:
Okay, with the correct settings the build continues almost till the end. But the build fails at the very end when generating the sdcard image:

```
cp /home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openmptcprouter-v0.59.2alpha-develop-8dc
d96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3-nor-bl31-uboot.fip /home/vbox/openmptcprouter/bpi-r3/5.15/source/bin/targets/mediatek/filogic/open
mptcprouter-v0.59.2alpha-develop-8dcd96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3-nor-bl31-uboot.fip
cat /home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openmptcprouter-v0.59.2alpha-develop-8d
cd96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.tmp >> /home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_corte
x-a53_musl/linux-mediatek_filogic/tmp/openmptcprouter-v0.59.2alpha-develop-8dcd96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
cp /home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openmptcprouter-v0.59.2alpha-develop-8dc
d96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3-snand-preloader.bin /home/vbox/openmptcprouter/bpi-r3/5.15/source/bin/targets/mediatek/filogic/ope
nmptcprouter-v0.59.2alpha-develop-8dcd96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3-snand-preloader.bin
rm /home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openmptcprouter-v0.59.2alpha-develop-8dc
d96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.tmp
dd if=/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openmptcprouter-v0.59.2alpha-develop-
8dcd96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz of=/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-
a53_musl/linux-mediatek_filogic/tmp/openmptcprouter-v0.59.2alpha-develop-8dcd96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new bs=1
7k conv=sync
DEVICE_ID="bananapi_bpi-r3" SOURCE_DATE_EPOCH=1671220716 FILE_NAME="openmptcprouter-v0.59.2alpha-develop-8dcd96b4-r0+21517-d7876daf65-mediatek-filogic-banana
pi_bpi-r3-emmc-preloader.bin" FILE_DIR="/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp" FI
LE_TYPE="emmc-preloader.bin" DEVICE_IMG_PREFIX="openmptcprouter-v0.59.2alpha-develop-8dcd96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3" DEVICE_VE
NDOR="Bananapi" DEVICE_MODEL="BPi-R3" DEVICE_VARIANT="" DEVICE_ALT0_VENDOR="" DEVICE_ALT0_MODEL="" DEVICE_ALT0_VARIANT="" DEVICE_ALT1_VENDOR="" DEVICE_ALT1_M
ODEL="" DEVICE_ALT1_VARIANT="" DEVICE_ALT2_VENDOR="" DEVICE_ALT2_MODEL="" DEVICE_ALT2_VARIANT="" DEVICE_TITLE="Bananapi BPi-R3" DEVICE_PACKAGES="kmod-hwmon-p
wmfan kmod-i2c-gpio kmod-sfp kmod-usb3 e2fsprogs f2fsck mkf2fs" TARGET="mediatek" SUBTARGET="filogic" VERSION_NUMBER="v0.59.2alpha-develop-8dcd96b4" VERSION_
CODE="r0+21517-d7876daf65" SUPPORTED_DEVICES="bananapi,bpi-r3" /home/vbox/openmptcprouter/bpi-r3/5.15/source/scripts/json_add_image_info.py /home/vbox/openmp
tcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/json_info_files/openmptcprouter-v0.59.2alpha-develop-8dcd96b4-r0+21517-d7876daf65-media
tek-filogic-bananapi_bpi-r3-emmc-preloader.bin.json
1+0 records in
1+0 records out
17408 bytes (17 kB, 17 KiB) copied, 0.000161472 s, 108 MB/s
mv /home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openmptcprouter-v0.59.2alpha-develop-8dc
d96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new /home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a5
3_musl/linux-mediatek_filogic/tmp/openmptcprouter-v0.59.2alpha-develop-8dcd96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
DEVICE_ID="bananapi_bpi-r3" SOURCE_DATE_EPOCH=1671220716 FILE_NAME="openmptcprouter-v0.59.2alpha-develop-8dcd96b4-r0+21517-d7876daf65-mediatek-filogic-banana
pi_bpi-r3-emmc-bl31-uboot.fip" FILE_DIR="/home/vbox/openmptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp" F
ILE_TYPE="emmc-bl31-uboot.fip" DEVICE_IMG_PREFIX="openmptcprouter-v0.59.2alpha-develop-8dcd96b4-r0+21517-d7876daf65-mediatek-filogic-bananapi_bpi-r3" DEVICE_
VENDOR="Bananapi" DEVICE_MODEL="BPi-R3" DEVICE_VARIANT="" DEVICE_ALT0_VENDOR="" DEVICE_ALT0_MODEL="" DEVICE_ALT0_VARIANT="" DEVICE_ALT1_VENDOR="" DEVICE_ALT1
_MODEL="" DEVICE_ALT1_VARIANT="" DEVICE_ALT2_VENDOR="" DEVICE_ALT2_MODEL="" DEVICE_ALT2_VARIANT="" DEVICE_TITLE="Bananapi BPi-R3" DEVICE_PACKAGES="kmod-hwmon
-pwmfan kmod-i2c-gpio kmod-sfp kmod-usb3 e2fsprogs f2fsck mkf2fs" TARGET="mediatek" SUBTARGET="filogic" VERSION_NUMBER="v0.59.2alpha-develop-8dcd96b4" VERSIO
N_CODE="r0+21517-d7876daf65" SUPPORTED_DEVICES="bananapi,bpi-r3" /home/vbox/openmptcprouter/bpi-r3/5.15/source/scripts/json_add_image_info.py /home/vbox/open
mptcprouter/bpi-r3/5.15/source/build_dir/target-aarch64_cortex-a53_musl/json_info_files/openmptcprouter-v0.59.2alpha-develop-8dcd96b4-r0+21517-d7876daf65-med
iatek-filogic-bananapi_bpi-r3-emmc-bl31-uboot.fip.json
[... cut ...]
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/vbox/openmptcprouter/bpi-r3/5.15/source'
make[1]: *** [target/Makefile:24: /home/vbox/openmptcprouter/bpi-r3/5.15/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/vbox/openmptcprouter/bpi-r3/5.15/source'
make: *** [/home/vbox/openmptcprouter/bpi-r3/5.15/source/include/toplevel.mk:231: world] Error 2
```

I have no idea how to fix this.

polskafan, 2023-01-12:
Got an image up and running on the eMMC of the R3. I am testing everything now. Launching v2ray causes a real high CPU load but OpenVPN and Wireguard seem to work. The upload speeds seem very slow for some reason, I need to investigate whats going on there. But in general it seems to be up and stable so far.

Andreadiv, 2023-03-18:
Hello. I have Banana PI R2 Pro. How can i compile for my version?

DominicMCN, 2023-04-20:
@polskafan Can you please share your config? I'm failing at that step as well

closed 2023-07-24 by github-actions[bot]