Ysurac/openmptcprouter #2596: Site to site IPSEC VPN connection

envlp, 2022-10-05:
## Expected Behavior

I need to connect site A and site B over a VPN connection
each site are equipped with a Stormshield firewall + OMR router + twice 4G (WAN1+WAN2) + VPS

## Current Behavior

I have try to connect site A to site B over IPSEC by setting the site firewall, but seems the aggregation work only on speedtest 

[image]

The questions are:
There are problems to use IPSEC ?
There are different way to connect the sites, for example directly by the OMR router configuration ?

## Specifications
  - OpenMPTCProuter version:  v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1028
  - OpenMPTCProuter VPS provider:Serverplan
  - OpenMPTCProuter platform: x86_64

envlp, 2022-10-06:
hello,
Thank you for the help!
Yesterday I saw this option on forum, I test it, but I was not able to establish the IPSEC connection again, I need to investigate on the IPSEC firewalls and the OMR port forwarding settings.
With this option enable, Iperf works fine with aggregation.
I will keep you informed

icolonder, 2022-12-10:
when i connect to a vpn, which is very close to the vps, it is too slow.