Ysurac/openmptcprouter #2601: Pip3 error on debian 11 install on hostworld VPS

drtechga, 2022-10-10:
Hello i keep getting error:
 user.notice OMR-VPS: Can't get hostworld2 token, try later (can ping server hostworld2 on  serveripaddress, no server API answer on  serveripaddress)
I have reinstall using the script multiple times and keep getting this error on the server during install:

ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
netjsonconfig 1.0.1 requires jinja2~=3.0.0, but you have jinja2 3.1.2 which is incompatible.
netjsonconfig 1.0.1 requires jsonschema~=4.4.0, but you have jsonschema 4.16.0 which is incompatible.

Ysurac, 2022-10-10:
Please use template when opening an issue.

As indicated the line before in log, I don't care about this error.
Did you reboot the VPS after running the script ? If yes, what is the result of "systemctl start omr-admin" ?

drtechga, 2022-10-10:
Yes I have restarted.

root@hwsopen1:~# systemctl start omr-admin
root@hwsopen1:~# systemctl status omr-admin
● omr-admin.service - OMR-Admin
     Loaded: loaded (/lib/systemd/system/omr-admin.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-10-10 01:24:06 EDT; 10min ago
   Main PID: 69660 (omr-admin.py)
      Tasks: 5 (limit: 2362)
     Memory: 42.6M
        CPU: 2.924s
     CGroup: /system.slice/omr-admin.service
             └─69660 /usr/bin/python3 /usr/bin/omr-admin.py

Oct 10 01:24:06  systemd[1]: Started OMR-Admin.

Ysurac, 2022-10-10:
What is the result of ```curl -k https://127.0.0.1:65500/``` on the VPS ? If you get an ouput, what is the result of ```curl -k https://<vpsip/:65500``` from the router ?

drtechga, 2022-10-10:
from VPS:
"Welcome to OpenMPTCProuter Server part"root@hwsopen1:~#

from router:
"Welcome to OpenMPTCProuter Server part"root@OpenMPTCProuter:~#

Here is also a port scan on the server:
PORT      STATE SERVICE

65001/tcp open  unknown

65101/tcp open  unknown

65222/tcp open  unknown

65228/tcp open  unknown

65301/tcp open  pcanywhere

65400/tcp open  unknown

65401/tcp open  unknown

drtechga, 2022-10-10:
It also seems to be a problem with hostworld as my OVH server connects fine.

drtechga, 2022-10-10:
I ended up finding the problem. I had a space in front of the wizzard server ip address setting. But it is wired that the log showed it could ping the server.

closed 2022-10-10 by drtechga