Ysurac/openmptcprouter #2602: OpenMPTCProuter degradated performance after few minutes

Death0rz, 2022-10-10:
## Expected Behavior

Expect that everything works like the first few mins
## Current Behavior

After 2/3 minutes from a reboot is stopping aggregate the wans (1 wisp connection 100/20mb, 1 5g of about 80-180/20-70mb) and the download speed is somehow limited to 10mb or less, on upload side there seems to not be limited but only use the first wan of 20mb. On first I thought some sort of limitation of my vm based on proxmox, but the same is happening on baremetal machine.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.59.1-5.15 r0+20029-3c06a344e9
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1028
  - OpenMPTCProuter VPS provider: ServerPlan IT
  - OpenMPTCProuter platform: x86_64 on a MiniPC Intel(R) Celeron(R) N5105 @ 2.00GHz with 4 Intel I226-V eths

Ysurac, 2022-10-10:
As you use testing 5.15 release, try to update VPS to 5.15 kernel, this may work better.

Death0rz, 2022-10-10:
Thanks for pointing me to that but issue is not resolved.
I'm getting 200/70mb on first speedtest tries and after a while back to 10/28mb.
Here a screen of the status and a pastebin of systemlog

https://pastebin.com/94edypdP

[image]

Ysurac, 2022-10-10:
Check the result of ```ip mptcp endpoint``` on both VPS and router via SSH.

Death0rz, 2022-10-10:
[image]

Ysurac, 2022-10-10:
Did you check in network->MPTCP "bandwidth" tab, that all interfaces are used during a speedtest ? (use ```omr-test-speed``` via SSH on the router)

Death0rz, 2022-10-10:
When is degraded nope, only the master wan1 connection work at this 10mb cap

Death0rz, 2022-10-10:
This entries on system log could be helpful?

`Mon Oct 10 20:49:38 2022 daemon.warn [6919]: <warn>  [SQM_IFB_2b779] invalid sysfs path read for net/SQM_IFB_2b779
Mon Oct 10 20:49:38 2022 daemon.warn [6919]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_2b779 not found
Mon Oct 10 20:49:38 2022 user.notice post-tracking-post-tracking: Set firewall on server vps
Mon Oct 10 20:49:38 2022 daemon.warn [6919]: <warn>  [SQM_IFB_8cd63] invalid sysfs path read for net/SQM_IFB_8cd63
Mon Oct 10 20:49:38 2022 daemon.warn [6919]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_8cd63 not found
Mon Oct 10 20:49:38 2022 daemon.err modprobe: failed to find a module named sch_fq_codel
Mon Oct 10 20:49:38 2022 daemon.warn [6919]: <warn>  [SQM_IFB_0db49] invalid sysfs path read for net/SQM_IFB_0db49
Mon Oct 10 20:49:38 2022 daemon.warn [6919]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_0db49 not found
Mon Oct 10 20:49:38 2022 daemon.err modprobe: failed to find a module named sch_fq_codel
Mon Oct 10 20:49:38 2022 daemon.warn [6919]: <warn>  [SQM_IFB_b89d5] invalid sysfs path read for net/SQM_IFB_b89d5
Mon Oct 10 20:49:38 2022 daemon.warn [6919]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_b89d5 not found`

Ysurac, 2022-10-10:
no, it's not related and not an error, only a normal warning.

closed 2023-01-14 by github-actions[bot]