Ysurac/openmptcprouter #2605: Shadowsocks is not running (empty key) and Can't connect with server, on RPI 4

atyouspace, 2022-10-11:
I get an error like the capture. What should I do?
[image]

I check that VPS admin key is good, that VPS was rebooted and I check State->System log.
[image]

Ysurac, 2022-10-11:
Please use the template when you open an issue.

Check that you rebooted the VPS after install, if yes check the result of ```systemctl start omr-admin``` on the VPS.

atyouspace, 2022-10-11:
> 
I reboot VPS, and check the results of `systemctl start omr-admin`, as followed,

root@iZ2zedyc2xu82zjw70o2s5Z:~# systemctl start omr-admin
Job for omr-admin.service failed because the control process exited with error code.
See "systemctl status omr-admin.service" and "journalctl -xe" for details.

Ysurac, 2022-10-11:
```pip3 install starlette```

Ysurac, 2022-10-11:
try ```pip3 install fastapi```

atyouspace, 2022-10-11:
> 

Its done. Why we need install fastapi?

Ysurac, 2022-10-11:
Because VPS API use FastAPI

hairy-dog, 2022-11-03:
I had this problem and wiing the vps then reistalling it fixed the problem.

closed 2023-02-06 by github-actions[bot]