Ysurac/openmptcprouter #2606: Wireguard VPN Client

freshwebs, 2022-10-11:
Hello folks,

I would like to connect directly from the OpenMPTCPRouter to my VPN via Wireguard.

For this purpose I added a new network adapter. This is called wg.

Protocol is Wireguard VPN.

I entered all the data there.

Then saved. But then comes the error: Network adapter is not available.

Why is this?

Ysurac, 2022-10-11:
Check Status->System log.

freshwebs, 2022-10-11:
Tue Oct 11 12:49:27 2022 daemon.notice netifd: Interface 'wg0' is setting up now
Tue Oct 11 12:49:27 2022 daemon.notice netifd: Interface 'wg0' is now down

freshwebs, 2022-10-11:
The Message "Error: Network adapter is not available." shows after adding content to tab "peers" from the wireguard adapter.

freshwebs, 2022-10-11:
Tue Oct 11 14:16:56 2022 daemon.notice netifd: Interface 'wg0' is setting up now
Tue Oct 11 14:16:56 2022 daemon.warn [5425]: <warn>  [wg0] invalid sysfs path read for net/wg0
Tue Oct 11 14:16:56 2022 daemon.warn [5425]: <warn>  [base-manager] couldn't handle kernel event: device net/wg0 not found
Tue Oct 11 14:16:56 2022 daemon.notice netifd: Interface 'wg0' is now down

freshwebs, 2022-10-27:
I think I found the option that is causing the problem. Unfortunately I don't have a solution yet.

As soon as I ticked "No host routes" in the Wireguard adapter, the connection does not start. The connection starts with the hook, but of course it doesn't do me much good because I want to reach the devices from the remote network.

closed 2023-03-22 by github-actions[bot]

MTSmhas, 2023-08-27:
I have the same problem. The openmptcrouter can't work as Wirguard Client. I think scrips for the Wireguard connection to the server destroy the session...