Ysurac/openmptcprouter #2610: Restore installed packages when backup is restored

Steltek, 2022-10-13:
I've installed a couple extra packages on my router (qrencode, keepalived, ...), however because I am running it as a VM, I do not have the option of running a sysupgrade and keeping installed packages that way (I have to drop the used HDD image and replace it with a new one instead).

It would be nice if the system could auto-install formerly installed/missing packages when the configuration is restored from backup. (The data seems to be in /etc/backup/installed-packages.txt, so it might just need a trigger or some extra scripting to do that installation after the reboot....)

Ysurac, 2022-10-13:
Should be the case, after an update installed packages are installed again if you update via the web interface.
I don't understand why you can't run a sysupgrade ?

Steltek, 2022-10-13:
There are no sysupgrade images for x86-64 (64-bits). I am running openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined.img.gz.

Ysurac, 2022-10-13:
You need to use this image to update via web interface. No need for a special sysupgrade image here.

closed 2023-01-17 by github-actions[bot]