Ysurac/openmptcprouter #2612: RUTX12 does not want to install

marccolemont, 2022-10-14:
This project is great. I very much appreciate your work. 
I bought a RUTX12, initiate to boot mode, see the recovery webpage on 192.168.1.1 => install the xxx.update.bin file for the RUTX12 (I tried multiple releases) 
After reboot / power cycle  no access on 192.168.100.1 on any RJ45 port. As if the firmware is not starting.

Looked around at this forum, and can't find what I'm I missing? And spend hours un uploading different versions.
Thanks for your help.

Ysurac, 2022-10-14:
Please complete template when you open an issue, and it's not the good category...
What is the tested image exactly ?
Lan port after reboot will be port labelled as WAN.

marccolemont, 2022-10-14:
Hi Ysurac. 
Sorry if this is the wrong place. I'm new here, spended whole afternoon on that RUTX12.
Yes I tried the WAN port after uploading (and all other ports)
I mostly tried the one from the download page: squashfs image: [openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-ipq40xx-generic-teltonika_rutx-squashfs-sysupgrade.bin](https://download.openmptcprouter.com/release/v0.59.1/rutx/targets/ipq40xx/generic/openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-ipq40xx-generic-teltonika_rutx-squashfs-sysupgrade.bin)
Or do I need a factory image first instead of an update image, but I don't see that available on the webpage like with the other hardware. 
Thanks very much for the help.

Ysurac, 2022-10-14:
It's the right image.
I just updated my RUTX12, so at least the image work (start take a quite long time). I will try to update from recovery webpage.
[image]

marccolemont, 2022-10-15:
I rolled back to the original RUTX firmware, upgraded then with the squashfs image and waited long enough. Works now. Thanks for the great support.

Ysurac, 2022-10-15:
Without needed to go to recovery webpage ?

marccolemont, 2022-10-15:
First In the recovery page I loaded the original RUTX12 image so the unit worked again. Then secondly I logged in and went to the firmware update and loaded the OpenMPTCrouter image from there.

Ysurac, 2022-10-15:
ok thanks, I will update documentation and add RUTX12 installation and that it's really a slow boot.

DylanPW, 2022-11-14:
Sorry for the thread necro, did the documentation page end up getting updated with the RUTX12 instructions? I remember previously seeing them in another issue thread but haven't seen it in the docs as of yet.

closed 2023-02-17 by github-actions[bot]

JohnTar22, 2023-04-18:
I know it's an old thread, tried to install the provided image on my rutx12, but it doesn't seem to work, what am i missing? tried reverting to stock firmware and upgrading from cli and ui but it just seem to halt on boot.

marccolemont, 2023-04-19:
Hi John, I never got the SIMS to detect and to work in bonding. Tried a zillion things, installed fresh multiple times, and did the steps described. Waiting for the next version to give it a try again. It's maybe a little setting somewhere. For some people the unit works.