Ysurac/openmptcprouter #2616: Applying a static lease on a device with wan1 ip range

FabriceC01, 2022-10-16:
## Expected Behavior
My new box player (Freebox Devialet) don't want to recognise the freebox server (192.168.2.1) with an IP given by OMR on the lan range (192.168.0.x). It only works when the Freebox server is DHCP and gives the player a 192.168.2.y IP server and gives the player a 192.168.2.y but there's conflict with the OMR DHCP. I've tried to configure the static lease in OMR DHCP to the plyer MAC with a 192.168.2.y IP but it doesn't works.

## Current Behavior
When I try to force an IP on lan range 192.168.0.x it's ok. the log is : 
_Sun Oct 16 20:41:08 2022 daemon.info dnsmasq-dhcp[6039]: DHCPDISCOVER(eth0) 70:fc:8f:61:a3:38
Sun Oct 16 20:41:08 2022 daemon.info dnsmasq-dhcp[6039]: DHCPOFFER(eth0) 192.168.0.158 70:fc:8f:61:a3:38
Sun Oct 16 20:41:08 2022 daemon.info dnsmasq-dhcp[6039]: DHCPREQUEST(eth0) 192.168.0.158 70:fc:8f:61:a3:38
Sun Oct 16 20:41:08 2022 daemon.info dnsmasq-dhcp[6039]: DHCPACK(eth0) 192.168.0.158 70:fc:8f:61:a3:38 PlayerCuisine_

When I try to force an IP on lan range 192.168.2.x it's as if ther's no static lease. the log is : 
_Sun Oct 16 20:35:03 2022 daemon.info dnsmasq-dhcp[7565]: DHCPDISCOVER(eth0) 70:fc:8f:61:a3:38
Sun Oct 16 20:35:03 2022 daemon.info dnsmasq-dhcp[7565]: DHCPOFFER(eth0) 192.168.0.143 70:fc:8f:61:a3:38
Sun Oct 16 20:35:03 2022 daemon.info dnsmasq-dhcp[7565]: DHCPDISCOVER(eth0) 70:fc:8f:61:a3:38
Sun Oct 16 20:35:03 2022 daemon.info dnsmasq-dhcp[7565]: DHCPOFFER(eth0) 192.168.0.143 70:fc:8f:61:a3:38
Sun Oct 16 20:35:03 2022 daemon.info dnsmasq-dhcp[7565]: DHCPREQUEST(eth0) 192.168.0.143 70:fc:8f:61:a3:38
Sun Oct 16 20:35:03 2022 daemon.info dnsmasq-dhcp[7565]: DHCPACK(eth0) 192.168.0.143 70:fc:8f:61:a3:38_

I can't specify an satic IP on the player. How can I force it on wan1 range with OMR ?

I've tried to active the freebox DHCP server with a static lease for the player and with an IP range of only 1. It's seems to work but I've had some strange events on my PC. I'm not sure it's linked.

PS : I've got some basic knowledges with lan but i'm not an expert.

Thanks for your help

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider: Milkywan
  - OpenMPTCProuter platform:

Ysurac, 2022-10-17:
You can use VLAN if you a manageable switch, manually set an IP in the 192.168.2.x range or provide an IP in 192.168.2.x via DHCP using the MAC address but I think you need to have also a LAN interface on OMR in this range.

closed 2023-01-20 by github-actions[bot]

flimo44, 2024-10-07:
Solution for me :
dhcp-option=tag:tag0,option:router,**192.168.2.1**
[image]

[image]

...
[image]

Then, I find that it's works and i can use now assistant.konodo . I hope these can help others!