Ysurac/openmptcprouter #2618: New install OS, being "readonly status" after power off

atyouspace, 2022-10-18:
## Expected Behavior
Edit file.

## Current Behavior
I installed OS on board, however, after power off and restart, almost everything become readonly status, such as /etc/config/network

## Specifications

  - OpenMPTCProuter version: 0.59
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: RPI4

winkmichael, 2022-10-18:
Just a guess, but have you tried using a different SD card? Sounds like maybe some disk corruption.

iambenmitchell, 2022-10-18:
ssh into the RPI and run this:

`/usr/sbin/e2fsck -y -v -f /dev/mmcblk0p2`

Then reboot.

Once rebooted, update the RPI's firmware with: rpi-eeprom-update -a

and reboot again. If you face any issues with updating the firmware check out the full article [here](https://github.com/Ysurac/openmptcprouter/wiki/Update-RPI4-firmware)

atyouspace, 2022-10-30:
When I use other platforms, the same problem is still there.
Such as imx6 series, readonly problem always appears. If I try `mount -o remount, rw /` the readonly problem could be solved temperately.
However, if I use openwrt official image, readonly problem vanishes. 
Is there any places in make menuconfig I should pay more attention?
@Ysurac

Ysurac, 2022-10-30:
imx6 are not supported.
But it's same problem with all cheap platform: when power is cut, last data in cache are not written on disk, so file system is corrupted.
You can try squashfs with lastest snapshot release, less data are written to disk.
Same problem will appear on OpenWRT images if you write files to disk...

highfreq, 2022-11-03:
Add the following to rc.local. It'll check status of the filesystem and if in readonly it'll fix it and reboot.

dane=$(mount | grep '/dev/root' | awk '{print $6}')
if [ "$dane" != "(rw,noatime)" ]
then
fsck.ext4 -y /dev/sda2

I run OMR from a usb stick so you may need to change /dev/sda2 to whatever your OS support is.

closed 2023-02-06 by github-actions[bot]