Ysurac/openmptcprouter #2620: Can't ping VPS server (RPI 3B)

kiryushah, 2022-10-19:
## Expected Behavior
A week ago everything was working fine. I didn't make any changes.

## Current Behavior
OpenMPTCPROUTER:  Shadowsocks is not running (empty key); VPN is not running; DNS Issue: cannot resolve hostname 
VPS: Cannot ping Server; Can't contact Server Admin Script (not token yet available)
Modems: No IP defined; No gateway defined 
I checked System Log: https://pastebin.com/t7iV46W6 
I checked VPS Server Key and Server Key in Setting Wizard - keys match
I restarted VPS, router, USB Hub - same trouble 
I  reinstall VPS and router images - same trouble

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: reg.ru
  - OpenMPTCProuter platform: RPI 3B

[image]

Ysurac, 2022-10-19:
The log is too short to say anything.
Check your WANs configuration as they have no IP/gateway set here. And the wan2 that use @wan1 interface seems strange.

kiryushah, 2022-10-19:
@Ysurac Thank you for your prompt response! After creating a case, I found a malfunction in the USB cable. Sorry for the false alarm.

closed 2022-10-19 by kiryushah