Ysurac/openmptcprouter #2621: 2 LTE modems - Rapsberry 4 - Router problem

KirillMilchakov, 2022-10-19:
## Expected Behavior
External router should be connected to Internet
## Current Behavior
Gateway down
## Specifications

  - [openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter VPS version: 0.59
  - OpenMPTCProuter VPS provider: reg.ru
  - OpenMPTCProuter platform: x86_64)

KirillMilchakov, 2022-10-19:
Hello, could anybody help me?
When I connect my OMR based on Rabsberry Internet is great, but when I use ethernet cable to connect with my router something going wrong( 
And One of my lte modems writes Gateway is down, and my router do not see internet connection at all.
please help!

rdmitry0911, 2022-10-20:
> Hello, could anybody help me? When I connect my OMR based on Rabsberry Internet is great, but when I use ethernet cable to connect with my router something going wrong( And One of my lte modems writes Gateway is down, and my router do not see internet connection at all. please help!

Can you show here the topology of your network when it works and when it does not.

KirillMilchakov, 2022-10-21:
Hello, everyone!
OpenMPTCP is already OK, OMR dynamically gives IP adress to Router and it works great.

But! I have noticed one thing! In case you have two LTE modems firstly you should connect Master eth port (in my case it is eth1), and only when it is connected plug the second lte modem. If to plug it together master modem shows GATEWAY IS DOWN. I thought it was router problem but no, it is because of sequence of connection.

rdmitry0911, 2022-10-21:
> But! I have noticed one thing! In case you have two LTE modems firstly you should connect Master eth port (in my case it is eth1), and only when it is connected plug the second lte modem.

This is probably because the system assigns names to the usb network interfaces depending on the order you plug them in. Check [this link ](https://forum.openwrt.org/t/rpi4-community-build/69998/1609) To fix this problem you have to install 000-kickanic script to /etc/hotplug.d/net/ and change your /etc/config/network according to the sample provided in the script. After that your usb network interfaces will have consistent names independent of the plugging in order.  
Here is the version of 000-kickanic script that perfectly works for me.
```

#!/bin/sh

#credits: bobafett vgaetera ...tba

############################################################################## this is how it works...

#####################################################step1 ### enable the logic as a whole
#uci set network.globals.persistent_nic='1'
#uci commit network

######################################################step2 ### for every usb nic
#uci set network.rummy.mac_original='00:00:e8:00:44:bf'
#uci commit network

################################################ update the uci logical interface underlying device eth+last6chars
#uci set network.rummy.device="eth0044bf"
#uci commit network

################# TIPS
### for extra wan interfaces repeat step2...#################################################################3caveats:

### first wan(all usb) need to be added
### NOTE: if you have wan + wan6 only use new device name on wan6 
### i.e. dont put identical mac_original twice (for now... still testing this)

### macaddr in device section will work if using fancy naming(picked up by netifd on rename)
### macaddr in interface section will work (picked up by this script/or os whichever changes first?)

### check /etc/config/sqm and/or /etc/config/luci_statistics(sqm_collectd) if the wan device name needs updating

################# CAVEATS
#no device bridges etc. for now
#usb only

###################################### EXAMPLE config format
#config globals 'globals'
#	option persistent_nic '1'

#config interface 'rummy'
#	option device 'eth0044bf'
#	option mac_original '00:00:e8:00:44:bf'
#	option proto 'dhcp'

#config interface 'wan'
#	option device 'ethdc03bf'
#	option mac_original '00:00:e8:dc:03:bf'
#	option proto 'dhcp'

		

nicmove_enabled() {
	local nmENABLED
	nmENABLED=$(uci -q get network.globals.persistent_nic)
	case "$nmENABLED" in
		1) return 0; ;;
		*) return 1; ;;
	esac
}

uci_nic_find() {

	local FN="uci_nic_find"
	local ucidevONE

[... 420 lines cut ...]

			usb)
				echV "INTERFACE:$INTERFACE $netdev_MAC_presented is mac_original unknown to /etc/config/network" msg
				echV "###################     WARNING     #################" msg
				echV "   ALL USB ADAPTERS MUST BE SETUP TO HANDLE BOOT     " msg
				echV "                    DETECTION PROPERLY               " msg
				echV "###################     WARNING     #################" msg

			;;

			*)
				#echV "INTERFACE:$INTERFACE $netdev_MAC_presented is mac_original unknown to /etc/config/network" debug
				echV "INTERFACE:$INTERFACE $netdev_MAC_presented is mac_original unknown to /etc/config/network" msg

			;;
		esac

		exit 0

	fi
fi

exit 0

```

Ysurac, 2022-10-21:
You can also try to uncheck the option "Disable interfaces auto rename" in System->OpenMPTCProuter, "Advanced settings" tab.

KirillMilchakov, 2022-10-21:
Great! thank you very much for help! And quick reaction:)

rdmitry0911, 2022-10-21:
> You can also try to uncheck the option "Disable interfaces auto rename" in System->OpenMPTCProuter, "Advanced settings" tab.

BTW, why don't you include this kickanic (or alike) script into OMR build? It is very common that people have multiple usb network adapters and using this script might make their config efforts easier

KirillMilchakov, 2022-10-21:
I will do it when I will be in office:) But, unfortunately I don't know how to add scripts to OMR manually. I installed packages by ssh before only(

KirillMilchakov, 2022-10-21:
How Can I upload files with scripts to OMR?

Ysurac, 2022-10-21:
> BTW, why don't you include this kickanic (or alike) script into OMR build? It is very common that people have multiple usb network adapters and using this script might make their config efforts easier

It's what should be done by the option I just gave...

KirillMilchakov, 2022-10-21:
Thank you @Ysurac I will try and come back with result.

KirillMilchakov, 2022-10-21:
And one more question: What does mean a message "File system is read only" near the OMR router on the page "OMR->Status"

Ysurac, 2022-10-21:
This mean that you have a problem.
You can check https://github.com/Ysurac/openmptcprouter/issues/2618

KirillMilchakov, 2022-10-25:
@Ysurac Great, it works)

KirillMilchakov, 2022-10-25:
But nevertheless I Have troubles with Gateway down problem still
[image]

Ysurac, 2022-10-25:
You should check if ping is allowed on these connections: try a "ping -b -I eth2 1.1.1.1", if you have no answer than ping is not allowed and you need to change settings in Services->OMR-Tracker.

KirillMilchakov, 2022-10-25:
Yes, ping is ok, and the servise is enabled

Ysurac, 2022-10-25:
Check Status->System log to know why eth2 and eth3 interfaces are disabled.

KirillMilchakov, 2022-10-25:
Tue Oct 25 10:54:34 2022 user.notice post-tracking-post-tracking: wan2 (eth2) switched off because check error and ping from 192.168.8.100 error (4.2.2.1,8.8.8.8,80.67.169.12)
Tue Oct 25 10:54:34 2022 user.notice post-tracking-post-tracking: Delete default route to 188.93.211.37 via 192.168.8.1 dev eth2

KirillMilchakov, 2022-10-25:
I didn't find lines about eth3 line at all

Ysurac, 2022-10-25:
Try "ping -B -I 192.168.8.100 188.93.211.37" via SSH from OMR router.

KirillMilchakov, 2022-10-25:
ping -b -I eth2 1.1.1.1 command is without reaction from OMR

KirillMilchakov, 2022-10-25:
In conclusion I have such picture:
[image]

KirillMilchakov, 2022-10-25:
no ping, no connection in modem B. If somebody has any ideas I would be grateful!)

Ysurac, 2022-10-26:
Can you ping this modem gateway from OMR via SSH ? ```ping -B -I 192.168.8.189 192.168.8.1``` ?
If not check modem IP configuration, if it's USB check that you have a powered USB hub,...

closed 2023-01-30 by github-actions[bot]