Ysurac/openmptcprouter #2626: Getting signal info and cell info from RUTX12

highfreq, 2022-10-27:
Anybody knows how i can get signal and carrier infos from the 2 modems inside the RUTX12?

highfreq, 2022-10-28:
If i use:  uqmi -d /dev/cdc-wdm0 --get-signal-info i get the info i need but the signal and other info disappear from the MPTCP web page system->openmptcprouter->status. Any other way to do it?

Ysurac, 2022-10-28:
By default ModemManager is used, so it's available with ```mmcli```

marccolemont, 2022-10-29:
Highfreq, did you get it to work? Have the same unit, configured SIMs with ModemManager, but it doesn't work here.

highfreq, 2022-10-29:
Yes, it works perfect. Didn't have to configure anything with modemmanager. What doesn't work on yours?

highfreq, 2022-10-29:
> By default ModemManager is used, so it's available with `mmcli`

Thanks, it worked!!!

closed 2023-02-01 by github-actions[bot]