Ysurac/openmptcprouter #2627: HDMI encoder not connecting to the internet when using OpenMPTCProuter

Pablosko98, 2022-10-27:
## Expected Behavior
Hi! I'm looking to feed an ethernet connection from a Raspberry Pi 4 running openmptcprouter to an HDMI encoder to bond several 4g connections together. 

## Current Behavior
Everything is working fine apart from the encoder, which is not able to connect to the internet.

I can't access the logs (at least that's what I think) since the raspberry is only connected to USB-tethered connections from a phone and a USB dongle to the encoder.
I've tried setting the encoder's gateway automatically with DHCP but that didn't work. I've also tried setting it to 192.168.1.254 and the same value for openmptc's router (instead of 192.168.100.1) manually which has also not worked.

For disclosure - the encoder works fine using public IPs etc, just not on openmptcprouter's network. Other devices (my laptop) are able to connect to the internet fine using this router.

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4
  - OpenMPTCProuter VPS version: (not sure how to find this)
  - OpenMPTCProuter VPS provider: AWS
  - OpenMPTCProuter platform: RPi 4 Model B Rev 1.4

Ysurac, 2022-10-28:
I don't understand your setup, can you make a diagram or something like that ?

Pablosko98, 2022-10-29:
Thanks for the reply, managed to solve it. Changing my video encoder's (connected to OpenMPTCProuter) IP from192.168.1.168 to 192.168.100.168 and its gateway to 192.168.100.1 seemed to work.
Just a couple of questions if you don't mind:

1) How can I effectively test that it is actually bonding the connections I'm giving it rather than just using one of them?
2) How long does it roughly take for a connection to join the "bonding" once plugged into the device?

Thank you in advance!

Ysurac, 2022-10-30:
Do  a speed test (for example with ```omr-test-speed``` via SSH on the router) and check in Network->MPTCP, "Bandwidth" tab

closed 2023-02-03 by github-actions[bot]