Ysurac/openmptcprouter #2628: Linksys WRT3200ACM won't go to internet , updated from v.0.58 - error warning in firmware upgrade

simonemessina92, 2022-10-28:
- OpenMPTCProuter version: v.0.59.1-5.4
  - OpenMPTCProuter VPS version: OpenMPTCProuter v0.59.1 
  - OpenMPTCProuter VPS provider: Aruba cloud
  - OpenMPTCProuter platform: Linksys WRT3200ACM
  - Country: Italy

Hello , I've 2 Linksys WRT3200ACM I had used with v.058 for a lot of time, yesterday I had tried to upgrade to v 0.59.1-5.4, first thing first I've got an error message while trying to upgrade the firmware, I've used the openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-mvebu-cortexa9-linksys_wrt3200acm-squashfs-sysupgrade.bin package in first place, after the reboot all the wan seems to work properly, getting the IP address from the LTE routers in DHCP, and got connection to every single router, but the router itself won't reach the internet in anyhow.
If I use the prompt of windows and try to ping a website I always got connection error, while trying to use the internal tool it seems it works, on windows I got the ethernet icon that seems alright, but indeed I can't reach anything in the network, only the router itself ofcourse is managable. 
[image]
It seems like I can't use the router with OMR, strange thing is that I've installed  the original firmware of open WRT and everything works fine , so it's not hardware related. I'm not super technical but it seems like the problem is that wan is not communicate to lan properly with the latest version of OMR, because can't reach the internet in anyhow, while in windows seems everything fine.
Weird thing I've notice is that at first boot after the firmware upgrade, all the wan were configured as static IP address, and after I swap them to DHCP the router itself reboot on its own, after the reboot every wan seems to work fine, got the local IP and the public IP address, but still not reach the internet. Wonder if there's as problem with the latest firmware version so far.
I've deploy 2 more mini pc with the latest OMR version and they work amazing though, so generally speaking I'm a afraid is firmware releated somehow.

thanks for the help in advance..

Ysurac, 2022-10-28:
Can you try to ping 1.1.1.1 ? to check if it's a DNS issue or not.

simonemessina92, 2022-10-29:
> Can you try to ping 1.1.1.1 ? to check if it's a DNS issue or not.

[image]
[image]

On OMR seems it works, but when I do on cmd in windows I've got all packet loss.

This is what happen when I try to flash the firmware from the standard open WRT ( which works btw )
[image]

After the first boot on OMR, I switch the first WAN to DHCP and this happen to the router, it basically reboot on it's own.

https://drive.google.com/file/d/1JpyhwGAbOuMV83j1MLeywSRgjIKUtleV/view?usp=sharing

Than after reboot the WAN1 seems ok , it got Ip address and on windows it seems to be connected, but not surfing the web...

[image]

I really don't know what the issue could be, but I've got 2 units on linksys 3200ACM and I wish I could use them again in my works, I've noticed that mini PC work really great for the OMR, but for convenience on having both the 4 rj45 and the 2USB on the back the Linksys are good for travelling and rental purpouses. 

Tell me If I can do more tests , I might attach the log and the kernel log here so you can see if there's something wrong somehow.
[log OMR.zip](https://github.com/Ysurac/openmptcprouter/files/9893091/log.OMR.zip)
[kernel log OMR.zip](https://github.com/Ysurac/openmptcprouter/files/9893092/kernel.log.OMR.zip)

Ysurac, 2022-10-29:
Check your client computer gateway and IP.
All interfaces on OMR seems to have same MAC address, I will check on my WRT3200ACM if it's the same...

simonemessina92, 2022-10-29:
> Check your client computer gateway and IP. All interfaces on OMR seems to have same MAC address, I will check on my WRT3200ACM if it's the same...

I usually leave everything on DHCP , so he default DSN is 192.168.100.1 as the gateway, I've also tried to force google's DNS but nothing changed so far. 
[image]
I just remember I've done another post in he past to solve that v2ray issue of stacking port forwarding with a specific ip address you've solved back in he days, and I was using the rc1 of 0.59 ( at least this was my last comment on that post ), than I thought I could easily upgrade to latest, because the mini pc worked amazing with 0.59.1, so if the problem is on firmware side you can update me on this, oherwise if you think i can use the vps server on 0.59.1 and the firmware on something diffrent I can give a try and install for example the rc3 of 0.95 which is the latest avaliable in the "previous" release in the download section.

Ysurac, 2022-10-29:
You can try latest snapshot: https://snapshots.openmptcprouter.com/5.4/wrt3200acm/targets/mvebu/cortexa9/
Do not keep config during update, do a fresh configuration.

simonemessina92, 2022-10-29:
> You can try latest snapshot: https://snapshots.openmptcprouter.com/5.4/wrt3200acm/targets/mvebu/cortexa9/ Do not keep config during update, do a fresh configuration.

I've done it right now, after the reboot the mac address were the same between the wans and after I've swap from static ip to dhcp the first wan the router rebooted itself as on previous version. 
[image]

I've done a fresh config as you said also

Ysurac, 2022-10-29:
Same mac are normal in fact.
Can you try to uncheck in Network->Firewall, the 2 offloading checkbox ?

simonemessina92, 2022-10-29:
> Same mac are normal in fact. Can you try to uncheck in Network->Firewall, the 2 offloading checkbox ?

[image]
Done it , but still unable to get internet access

simonemessina92, 2023-01-05:
@Ysurac  I've tried to flash over and over again the firmware on my 2 linksys unit but nothing happen, still won't able to make them work with the last release. I've tested both with openwrt v.23 and they seems to work fine, I was able to also add load balancing, but of course is nothing compared to OMR... Do you think is somehow firmware related or there's some more advanced way to install OMR rather than flash the system through the luci interface?  just the fact that no matter from which version I flash this last release I'll always got the error that file does not contain a supported format it seems really strange to me... I don't know what to do, but I don't want to trash out 2 routers, I rather prefer to understand how to solve this. I can defintily try to do whatever you can suggest on this. The weird thing is that it seems like the wan connection got dhcp address and are uptime, but than won't comunicate through the lan zone... it's really weird. On normal x64 platforms it v.0.59.1 works really great ...
[image]

Ysurac, 2023-01-05:
What is the file used ?

simonemessina92, 2023-01-05:
> What is the file used ?

I'm always downloading the latest file from the site https://www.openmptcprouter.com/download so the last one, I've also tried the one you've sent me in the link up top here, but still the same problem. the router doesn't  surf the web anyhow, neither using one only connection without the server active... As I said I don't know if there's a sort of way to erase everything in the linksys and manually force to do some sort of clean installation... the point is that now I've 2 routers that won't work at all, and I really don't want to put it in the corner, I'm pretty sure this situation can be solved somehow..

Ysurac, 2023-01-05:
I've already tested current release on WRT3200ACM and update is working (without theses warning). You can also update via terminal https://github.com/Ysurac/openmptcprouter/issues/2629

For the update via luci or OpenWRT command line, the sysupgrade.bin image must be used.
You can specify in both case if you want to keep or erase current settings.

simonemessina92, 2023-01-05:
> I've already tested current release on WRT3200ACM and update is working (without theses warning). You can also update via terminal #2629
>
> For the update via luci or OpenWRT command line, the [...]

I've actually had tried to both systemupgrade.bin ( from openwrt and omr ) and also the factory one, it always gave me that error, and once it flash on the linksys there's no way to make it work. That's why I'm trying to get a diffrent approach from you , I always unchecked the keep the current setting. I well had understand how to properly settlle that at this point, there's not the really problem. The issue is that no matter what or how I flash it it end it up to not work, and from whatever version I try it always gave me that top screenshot error. Is it possibile there's no way to troubleshoot this issue ?

Ysurac, 2023-01-05:
You correctly tested using the Internet port for LAN ?
What are the led result after upgrade ?

simonemessina92, 2023-01-05:
> You correctly tested using the Internet port for LAN ? What are the led result after upgrade ?

for sure I'm using the correct port for the lan otherwise I wouldn't even get an Ip address from the router, once I flash the firmware as always the wan port become to be the LAN port, and the 4 lan port are the wan, even in the dashboard it seem everything is fine, but it actually won't surf the web anyhow, can't do a single ping from the cmd line of windows, or even reach the ip of the wan gateways. while if I attach the single external router antenna it works and I can easily ping the single router one by one and also use the internet connection . The strange thing is that in the dashboard it read the public ip address of the first wan connection but still won't work.
[image]
[image]
[image]

Ysurac, 2023-01-05:
Can you try to disable offloading in firewall->general setting tab ?

simonemessina92, 2023-01-05:
> Can you try to disable offloading in firewall->general setting tab ?

Done it, it was something we've tried to do it when I've opened the post but notthing changed so far :(

simonemessina92, 2023-01-05:
@Ysurac Is there maybe something that had to be changed into the firewall tab ? I mean it really seems like the lan zone won't communicate with the wan zone. How does it possible to not even reach the default gateway of the first wan connection router attached on the Linksys ? In the interface I can clearly see the external router gave the wan 1 ip address, but if i point to ping or even via http to the default gateway of the external connection I can't reach it. I've another OMR built with the same release but on x64 plaform on a "simil" nuc PC... I can clearly reach every gateway of the wan connection if I use that OMR. I'm really struggling to understand what the problem could be with this Linksys hardware I have... It really seems like there's no interaction between the two zones I can't explain differently, I'm not an IT expert, I well known this, but I work with networks everyday and I'm far away to be in the top of the noobs charts😂 the thing that I'm suffering the most is that even watching carefully the menu's one by one everything seems as it should be... I repeat , idk if there's some other advanced terminal way to investigate this problem to the root of this problem, I'm asking to you that are the big expert here 🙏

simonemessina92, 2023-01-06:
@Ysurac  Now I'm trying accessing via SSH to the OMR, and actually from there I'm able for example to ping the gateway of my first wan connection. But if I try to open it on the browser and reach the wan1 gateway it doesn't load the page, and in the dashbord I got no connection at all; even if the 3 wan I've got connected now got their own dhcp address, and they work correctly If I remove the OMR from the chain. If I attach the external lte antenna to the pc I got connection and so on, so I can clearly say it's not a problem of the wan connection not working, I've also tried with my home ftth line and it works of course, but it seems like Wan zone won't pass through lan zone and let me surf the web. I've also check to compare the x86 OMR I got with this linksys model, and the menu are pretty much indentical... screenshot attached
[image]
[image]
[image]
[image]

simonemessina92, 2023-01-06:
Ok this is really weird, I've always tested in normal mode without enabling the bonding in the MPTCP tab... but when I enable the connection to the vps server everything seems to work... That's the most akward thing I've seen on this OMR platform. The x86 variant work also without the server bonding active... it seems that as soon I've enabling the server connection at least it surf the web. on the speed test I've got the ip address of the cloud vps. Still wondering why I can't ping the gateway of the wan connections by the way... It seems working but there's something still not, on the x86 platform I can easily see the wan gateways... there's is still some complains about the working on this release ...
[image]

blackdragoncn, 2023-03-06:
In my two Linksys WRT32Xs, there is the same problem. The VPS server is good, the OMR router is good, all the clients can use internet as usual, But the Ping comand can't work in client's PC, what ever IP address be. 
The Image is "openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-mvebu-cortexa9-linksys_wrt32x-squashfs-sysupgrade.bin", now I will try to use the new one "openmptcprouter-v0.59.2alpha-5.4-r0+16819-1b6e9b3f64-mvebu-cortexa9-linksys_wrt32x-squashfs-sysupgrade.bin".

blackdragoncn, 2023-03-06:
The new image has the same problem......

closed 2023-06-10 by github-actions[bot]