Ysurac/openmptcprouter #2629: Linksys WRT32X - Firmware upload failed

n4sty3us, 2022-10-29:
## Expected Behavior
Hello, I've Linksys WRT32X on which I want to install the firmware OpenMPTCProuter. I downloaded the latest version available for my router from the website: openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-mvebu-cortexa9-linksys_wrt32x-squashfs-factory.img

## Current Behavior
When uploading the image, I get the following error: Firmware upload failed

[image]

Could I have information on how to solve the problem? Thank you!!!

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4
  - OpenMPTCProuter platform: Linksys WRT32X

Ysurac, 2022-10-31:
You can try to update from OpenWRT (without keeping settings).
As firmware upload failed, this may be because there is not enough free space.

n4sty3us, 2022-10-31:
@Ysurac thank you so much for the support!!! If I understand correctly, I update the OEM firmware with OpenWRT. After I update the OpenWRT firmware with the OpenMPTCProuter firmware ... is it correct?

I have two questions for you:
- To switch from OpenWRT to OpenMPTCProuter can be done from GUI or command line?
- Which image should I use? (openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-mvebu-cortexa9-linksys_wrt32x-squashfs-factory.img or openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-mvebu-cortexa9-linksys_wrt32x-squashfs-sysupgrade.bin)

Thanks

closed 2022-10-31 by n4sty3us