Ysurac/openmptcprouter #2632: LTE Gateway DOWN

NKnusperer, 2022-10-31:
## Expected Behavior
I'm using a Sierra Wireless MC7455 LTE (QMI mode) card via an USB enclosure on a PI4.
When establish a connection via ModemManager a IP is assigned from my carrier and the route should be UP.

## Current Behavior
Route is marked as DOWN.
The LTE link itself seems to be usable, eg. `omr-test-speed wwan0` works as expected.
Also tried version `0.59.2alpha-54` but the behavior is the same as with the current stable release.
Maybe related: When doing the initial setup using the Wizard the LTE device is somehow not properly selected.
Under `Network` -> `Interfaces` the ModemManager reports an error: `Network device not present`.
After clicking `Edit` and `Save` the connection is suddenly established using the device.

[image]

```
root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='5'
network.globals=globals
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.congestion='cubic'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_subflows='3'
network.globals.mptcp_add_addr_accepted='1'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.ula_prefix='fd59:0c8c:4825::/48'
network.globals.mptcp_syn_retries='2'
network.lan=interface
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.device='eth0'
network.lan.ifname='eth0'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.addlatency='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.metric='6'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.ip4table='wan'
network.wan1.multipath='master'
network.wan1.defaultroute='0'
network.wan1.addlatency='0'
network.wan1.ipv6='0'
network.wan1.metric='3'
network.wan1.peerdns='0'
network.wan1.proto='dhcp'
network.wan1.device='eth1'
network.wan1_dev=device
network.wan1_dev.txqueuelen='20'
network.wan1_dev.name='eth1'
network.wan2=interface
network.wan2.ip4table='wan'
network.wan2.multipath='on'
network.wan2.defaultroute='0'
network.wan2.addlatency='0'
network.wan2.ipv6='0'
network.wan2.metric='4'
network.wan2.peerdns='0'
network.wan2.proto='modemmanager'
network.wan2.apn='nonbonding.hybrid'
network.wan2.device='/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3'
network.wan2.auth='none'
network.wan2.iptype='ipv4'
network.wan2_dev=device
network.wan2_dev.txqueuelen='20'
network.wan2_dev.name='wan2'
network.omrvpn=interface
network.omrvpn.device='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.type='tunnel'
network.omrvpn.txqueuelen='100'
network.omrvpn.metric='1200'
network.omrvpn.proto='none'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.auto='0'
network.omr6in4.metric='1201'
network.omr6in4.ip6addr='fe80::a00:2/126'
network.omr6in4.gateway='fe80::a00:1/126'
network.lan_dev=device
network.lan_dev.name='eth0'
```

## Specifications

  - OpenMPTCProuter version: 0.59.2alpha-54
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: RPI4

Network-Traditions, 2022-10-31:
@NKnusperer just thought I'd mention that our 5G T-Mobile USA service blocks gateway ICMP pings.  If we leave "Disable gateway ping" unchecked, OMR reports that the gateway is "DOWN" even though aggregation is working.  Disabling the gateway ping eliminates this check and therefore the resulting warning message on the status dashboard.

NKnusperer, 2022-10-31:
Thank you @Network-Traditions, this explains the observed behavior.
Is it possible to use a custom endpoint for monitoring (eg 8.8.8.8)?

Network-Traditions, 2022-11-01:
Not having knowledge regarding the internal functionality of OMR I am unable to affirmatively confirm the following, however, months of testing and close monitoring of our deployment has led me to believe that OMR pings or otherwise establishes connections with a number of internet hosts (see Services-OMR-Tracker).  Governed by the "Other settings" section of System-OpenMPTCProuter-Advanced Settings options like Disable gateway ping, Disable server http test, VPS checks timeout, WAN IPs retrieve timeout, Disable server ping, Disable route loop detection, Disable tracebox test and Disable nDPI all govern the monitoring and warning process for OMR.  Subsequently, should a WAN connection fail to properly pass traffic to and from the internet in general, these tracking services reveal the failure and the dashboard provides a message stating a problem specific to the connection that is not working properly.  Additionally, "latency" feedback for the WAN interfaces is provided on the System-OpenMPTCProuter-Status page.  I find these numbers of higher value to provide configuration and operational quality guidance.  As latency exceeds 300ms and their deltas between the WAN interfaces increases, the quality of the aggregated connection decreases.

closed 2023-02-05 by github-actions[bot]