Ysurac/openmptcprouter #2634: Provide minicom package

NKnusperer, 2022-11-02:
**Is your feature request related to a problem? Please describe.**
Most LTE modem cards support AT commands to communicate directly with the device for low-level configuration via an exposed serial port (`/dev/ttyUSB*`). To my knowledge OMR currently does not provide any terminal emulator to communicate with such an interface except using `socat` or plain `echo` which is not really a nice experience.

**Describe the solution you'd like**
Include minicom with the base OMR distribution.
https://openwrt.org/packages/pkgdata/minicom

Ysurac, 2022-11-02:
minicom is already available to install via System->Software.

NKnusperer, 2022-11-05:
True, I did not updated the software list first.

closed 2022-11-05 by NKnusperer