Ysurac/openmptcprouter #2635: SSH and Luci not responding on ipv6 after enabling VPS/bonding

QCWBradleyGebhardt, 2022-11-03:
## Expected Behavior
We should be able to shh via ipv6 as  well as load up the web interface on ipv6

## Current Behavior
The ssh session is immediately closed by the remote:
```bash
$ ssh user@[ipv6 address]
  kex_exchange_identification: Connection closed by remote host
  Connection closed by [ipv6 address] port 22
```

and the web interface doesn't send any data:
```
This page isn’t working
[ipv6 address] didn’t send any data.
ERR_EMPTY_RESPONSE
```

These two things only happen when the VPS is enabled. Otherwise it works just fine.

## Specifications

  - OpenMPTCProuter version: 0.59-5.4 and 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: AWS Debian 11
  - OpenMPTCProuter platform: x86

Ysurac, 2022-11-07:
I need more details. IPv6 are public IP or ULA ? You try from another local client ?

QCWBradleyGebhardt, 2022-11-08:
The v6 address we are using have a prefix of fc00:aa:aa::/48 and fc00:aa:aa:221:1::/80. And I tried pinging from another local client. Hope that helps

closed 2023-02-12 by github-actions[bot]