Ysurac/openmptcprouter #2636: DHCPV6 and IPv6 not working when wan status changes

datapharmer, 2022-11-03:
## Expected Behavior

Connecting a client via dual stack ipv4 and ipv6 or just ipv6 should result in getting an ipv6 address issued with valid gateway
## Current Behavior

Upon first starting or restarting LAN interface clients can get a dhcpv6 address issued that is within the range found listed under wizard > IPv6 settings and ipv6 gateway and dns servers are included and functional. Everything works. After a few minutes any clients trying to lease or renew a lease only get IPv4 if dual stack or if IPv6 only they get a link local address no gateway and invalid dns such as fec0:0:0:ffff::1%1 and clients that already have a lease stop passing ipv6 traffic (see ping example below)
## Possible Solution

Restart the LAN connection if status of one of the other wan interfaces changes?.... this seems a bit disruptive
## Steps to Reproduce the Problem

  1. Restart lan
  2. renew dhcp on client
  3. confirm ipv6 is working by reviewing address and using ping -6 google.com and that lease appears in ui under dhcp and dns > static leases > active dhcpv6 leases
  4. wait for one of the wan connections to become unavailable
  5. Note that no valid IPv6 is issued and ipv6 client traffic starts to fail

## Context (Environment)

dual wan set to restart interface if it is detected as down. 

Ping on IPv4 works reliably, but here is an example of a long ping running IPv6 from a dhcp client:
```
Reply from 2607:f8b0:4008:809::200e: time=83ms
Reply from 2607:f8b0:4008:809::200e: time=79ms
Reply from 2607:f8b0:4008:809::200e: time=66ms
Reply from 2607:f8b0:4008:809::200e: time=87ms
Reply from 2607:f8b0:4008:809::200e: time=72ms
Reply from 2607:f8b0:4008:809::200e: time=80ms
Reply from 2607:f8b0:4008:809::200e: time=87ms
Reply from 2607:f8b0:4008:809::200e: time=314ms
Reply from 2607:f8b0:4008:809::200e: time=120ms
Reply from 2607:f8b0:4008:809::200e: time=147ms
Reply from 2607:f8b0:4008:809::200e: time=395ms
Request timed out.
Reply from 2607:f8b0:4008:809::200e: time=381ms
Reply from 2607:f8b0:4008:809::200e: time=291ms
Reply from 2607:f8b0:4008:809::200e: time=198ms
Request timed out.
PING: transmit failed. General failure.
PING: transmit failed. General failure.
PING: transmit failed. General failure.
PING: transmit failed. General failure.
PING: transmit failed. General failure.
PING: transmit failed. General failure.
PING: transmit failed. General failure.
PING: transmit failed. General failure.
PING: transmit failed. General failure.
PING: transmit failed. General failure.
PING: transmit failed. General failure.
PING: transmit failed. General failure.
```
I'm also seeing an IPv6 related error in the logs though it doesn't seem to be directly related to when the failures start as it happens before and after as well:
Thu Nov  3 15:12:41 2022 kern.err kernel: [2119543.645889] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Thu Nov  3 15:12:41 2022 kern.err kernel: [2119543.654023] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Thu Nov  3 15:12:42 2022 kern.err kernel: [2119545.453091] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Thu Nov  3 15:12:42 2022 kern.err kernel: [2119545.462228] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Thu Nov  3 15:12:42 2022 kern.err kernel: [2119545.470133] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Thu Nov  3 15:12:42 2022 kern.err kernel: [2119545.478266] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Thu Nov  3 15:12:43 2022 kern.err kernel: [2119546.404484] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13

## Specifications

  - OpenMPTCProuter version:   v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version:  0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: BuyVM
  - OpenMPTCProuter platform:  RPI 4 64-bit
  - Country:  U.S.

Ysurac, 2022-11-03:
This may be a firewall config not correctly reloaded.
Can you give me the result of ```ip6tables-save``` before and after problem ?

datapharmer, 2022-11-04:
before:

```
# Generated by iptables-save v1.8.7 on Thu Nov  3 18:42:32 2022
*raw
:PREROUTING ACCEPT [546574100:353900767434]
:OUTPUT ACCEPT [461266778:287577660933]
COMMIT
# Completed on Thu Nov  3 18:42:32 2022
# Generated by iptables-save v1.8.7 on Thu Nov  3 18:42:32 2022
*nat
:PREROUTING ACCEPT [18479:2289361]
:INPUT ACCEPT [11384:719173]
:OUTPUT ACCEPT [42326:3169870]
:POSTROUTING ACCEPT [4233:254410]
:MINIUPNPD - [0:0]
:MINIUPNPD-POSTROUTING - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_vpn_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_vpn_rule - [0:0]
:prerouting_wan_rule - [0:0]
:v2r_def_dst - [0:0]
:v2r_def_forward - [0:0]
:v2r_def_local_out - [0:0]
:v2r_def_pre_src - [0:0]
:v2r_def_src - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_vpn_postrouting - [0:0]
:zone_vpn_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -i eth0 -m comment --comment "!fw3" -j zone_lan_prerouting
													
-A PREROUTING -m comment --comment "!fw3: Custom prerouting rule chain" -j prerouting_rule
-A PREROUTING -i eth0.3 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i eth0.2 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i tun0 -m comment --comment "!fw3" -j zone_vpn_prerouting
-A PREROUTING -i 6in4-omr6in4 -m comment --comment "!fw3" -j zone_vpn_prerouting
-A PREROUTING -p tcp -j v2r_def_pre_src
-A OUTPUT -p tcp -j v2r_def_local_out
-A POSTROUTING -m comment --comment "!fw3: Custom postrouting rule chain" -j postrouting_rule
-A POSTROUTING -o eth0 -m comment --comment "!fw3" -j zone_lan_postrouting
-A POSTROUTING -o eth0.3 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o eth0.2 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o tun0 -m comment --comment "!fw3" -j zone_vpn_postrouting
-A POSTROUTING -o 6in4-omr6in4 -m comment --comment "!fw3" -j zone_vpn_postrouting
-A v2r_def_dst -m set --match-set omr_dst_bypass_all dst -j MARK --set-xmark 0x539/0xffffffff
-A v2r_def_dst -m mark --mark 0x539 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_eth0_2 dst -j MARK --set-xmark 0x53910/0xffffffff
-A v2r_def_dst -m mark --mark 0x53910 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_eth0_3 dst -j MARK --set-xmark 0x5398/0xffffffff
-A v2r_def_dst -m mark --mark 0x5398 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_6in4-omr6in4 dst -j MARK --set-xmark 0x5391201/0xffffffff
-A v2r_def_dst -m mark --mark 0x5391201 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_tun0 dst -j MARK --set-xmark 0x5391200/0xffffffff
-A v2r_def_dst -m mark --mark 0x5391200 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_eth0 dst -j MARK --set-xmark 0x5396/0xffffffff
-A v2r_def_dst -m mark --mark 0x5396 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_lo dst -j MARK --set-xmark 0x5395/0xffffffff
-A v2r_def_dst -m mark --mark 0x5395 -j RETURN
-A v2r_def_dst -m set --match-set ss_rules_dst_bypass_all dst -j RETURN
-A v2r_def_dst -m set --match-set ssr_def_dst_bypass dst -j RETURN
-A v2r_def_dst -m set --match-set ssr_def_dst_forward dst -j v2r_def_forward
-A v2r_def_dst -m comment --comment "dst_default: forward" -j v2r_def_forward
-A v2r_def_forward -p tcp -j REDIRECT --to-ports 1897
[... 728 lines cut ...]
-A zone_vpn_src_REJECT -i 6in4-omr6in4 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_ACCEPT -o eth0.3 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o eth0.3 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_ACCEPT -o eth0.2 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o eth0.2 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_REJECT -o eth0.3 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o eth0.2 -m comment --comment "!fw3" -j reject
-A zone_wan_forward -m comment --comment "!fw3: Custom wan forwarding rule chain" -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "!fw3: Allow-IPSec-ESP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "!fw3: Allow-ISAKMP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port forwards" -j ACCEPT
-A zone_wan_forward -m comment --comment "!fw3" -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "!fw3: Custom wan input rule chain" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment "!fw3: Allow-DHCP-Renew" -j ACCEPT
-A zone_wan_input -p icmp -m icmp --icmp-type 8 -m comment --comment "!fw3: Allow-Ping" -j ACCEPT
-A zone_wan_input -p igmp -m comment --comment "!fw3: Allow-IGMP" -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port redirections" -j ACCEPT
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j output_wan_rule
-A zone_wan_output -m comment --comment "!fw3" -j zone_wan_dest_ACCEPT
-A zone_wan_src_REJECT -i eth0.3 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i eth0.2 -m comment --comment "!fw3" -j reject
COMMIT
# Completed on Fri Nov  4 10:57:30 2022
```

datapharmer, 2022-11-09:
@Ysurac any thoughts? this is causing me some big headaches because dns and connections time out/fail most of the time when ipv6 is enabled ont he client.

datapharmer, 2022-11-09:
Should OMR6IN4 and WAN connections have `Delegate IPv6 prefixes` enabled? It is enabled for OMR6IN4 but thinking maybe that is the problem since I'm seeing it set a default route on the client and seems like sometimes it isn't passing traffic and sometimes it is.

datapharmer, 2022-11-11:
Ok I've found some of the problems here and I think have it mostly fixed.

OMR6IN4 didn't have an IPv6 assignment length set and this was being used for DHCP assignment (not sure that's right, but that's what's happening) so only one device at a time was getting an address.

Once I changed that and set Delegate IPv6 prefixes on for OMR6IN4 and the wan connections everyone is getting a valid IPv6 address and generally traffic is passing through the tunnel. However occasionally I'm still seeing IPv6 stop routing which you can also see on the router as it can't ping anything outside the lan either. I tried restarting interfaces etc. to no avail for this scenario. A full reboot brings it back to working so I think this may be a firewall issue or problem with v2ray?

I did not this in the logs which seems to point to an issue with OMR6IN4:

```
Fri Nov 11 14:52:26 2022 kern.err kernel: [  847.767702] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Fri Nov 11 14:52:27 2022 kern.err kernel: [  848.620114] __mptcp_init6_subsockets: token 0x78d93d7d bind() to [OMR6IN4 Interface IP] index 22 failed, error -99
```

datapharmer, 2022-11-11:
Ok if I reload the firewall from the cli it gives this:

datapharmer, 2022-11-11:
* Populating IPv6 nat table
Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_lan_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_lan_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_wan_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_wan_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_vpn_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_vpn_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_rule'

This let me to this known bug that was never addressed: https://github.com/openwrt/openwrt/issues/8534
But I found another reference here: https://forum.openwrt.org/t/firewall-gives-warning-msgs-when-kmod-ipt-nat6-is-installed/71285/2

Which suggests this as a potential fix: https://openwrt.org/docs/guide-user/network/ipv6/ipv6.nat6#nat6_simplified
but would it be better to run relay mode or stuck with server mode and use this nat fix?

datapharmer, 2022-11-18:
I've determined this is a problem with route announcement. It seems the ipv6 routes are expiring and not renewing. Restarting the network brings them back for a bit then they expire again. 

@Ysurac Any suggestion on how to get a new ra or something to prevent them from being withdrawn? I tried the openwrt allmulti suggestion related to vlan (https://github.com/openwrt/openwrt/issues/9827) as I thought this would apply to my scenario, but it doesn't seem to have solved it. 

I can see that the routes to omr6in4 seems to get lost and the ipv6 dns server is missing from dhcp lease when this happens. Once the network is restarted and a lease renewal requested it returns to normal for a little while until the ra gets lost again. 

This is just a problem for the dhcp clients as the router can still ping an ipv6 address.

I've tried running the lan in both server and with pd/relay mode and experience the same route loss regardless

kevinh-csalabs, 2022-11-23:
After a ton of digging I found that setting noserverunicast on the wan connections seems to be the solution:
`option noserverunicast '1'`
https://blog.printk.io/2018/08/ipv6-renew-issue-with-fiber7-and-openwrt/

After setting this I've had ipv6 working for a few days with the exception if the vpn tunnel is down. When that's the case ipv6 ping doesn't work on clients or from the router, but that appears to be a separate bug.

Leaving this open for now as it might be good to set the noserverunicast option by default as it isn't immediately apparent things aren't working unless you happen to check after things have been running for a while.

kevinh-csalabs, 2022-12-30:
Ok, so follow up that this isn't a total fix, but I know another bug was found and fixed in dev but I don't have a means of testing that.

closed 2023-04-04 by github-actions[bot]