Ysurac/openmptcprouter #2639: OpenMPTCP : No data error

hamzaaboufirass, 2022-11-04:
I am on openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c

everything is configured correctly , 3 wan lines (DHCP) + 1 lan (bridge over two ports) lan has dhcp enabled.

When connecting to VPS i get a No Data error on the status page and no internet is available:
[image]

[image]

Is this a bug?

Ysurac, 2022-11-04:
Please use template when you open an issue.
Try to do a fresh configuration without any bridge.

hamzaaboufirass, 2022-11-04:
Hello Ysurac,

My apology for the template, didn't pay attention to it.

Tried without bridge still same issue.

I have uploaded the log as well.

[log-mptcp.zip](https://github.com/Ysurac/openmptcprouter/files/9941514/log-mptcp.zip)

Network-Traditions, 2022-11-09:
I had this issue when I modified the default LAN configuration of OMR (see #2540).  After I identified the problem with changing LAN naming, I rebuilt OMR without modifying the default LAN (eth0) setup and the "No data" issue was resolved.  Due to my limited experience and knowledge of OMR/OpenWRT configuration, I'm guessing changing the LAN name requires additional modification of firewall rules and/or other impacted settings.  Not wanting to waste time identifying those issues, I opted for a clean build with the default LAN setup.

closed 2023-02-13 by github-actions[bot]