Ysurac/openmptcprouter #2642: Impossible de contacter le script administrateur (no token yet available)

Copinmalin, 2022-11-06:
## Expected Behavior

Everything seems working, but I have this issue on VPS view : 

Impossible de contacter le script d'administration du serveur (no token yet available)
[image]

## Current Behavior

System log :  

`OMR-VPS: Can't get vps token, try later (can ping server vps on 51.xxx, server API answer on 51.xxx)`

API Key exist in VPS config

## Specifications

  - OpenMPTCProuter version:  0.59.1-5.4
  - OpenMPTCProuter VPS version:  0.1028
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.4

Ysurac, 2022-11-06:
Check that you set correctly API key and that it is the same in /etc/openmptcprouter-vps-admin/omr-admin-config.json

Copinmalin, 2022-11-06:
Many thanks @Ysurac , you are a boss 

server key was configured with API keys !

closed 2022-11-06 by Copinmalin