Ysurac/openmptcprouter #2643: glory tun interface is toooo slow. Why?

rdmitry0911, 2022-11-07:
Hi guys, I have 3 links and all of them have quite normal ping time  about 100ms (see attached), however the glorytun (tun0) has huge ping time about 1100ms like this:

```
root@OpenMPTCProuter:~# ping -c 10 10.255.255.1
PING 10.255.255.1 (10.255.255.1) 56(84) bytes of data.
64 bytes from 10.255.255.1: icmp_seq=1 ttl=64 time=656 ms
64 bytes from 10.255.255.1: icmp_seq=2 ttl=64 time=2547 ms
64 bytes from 10.255.255.1: icmp_seq=3 ttl=64 time=1636 ms
64 bytes from 10.255.255.1: icmp_seq=4 ttl=64 time=1066 ms
64 bytes from 10.255.255.1: icmp_seq=5 ttl=64 time=1048 ms
64 bytes from 10.255.255.1: icmp_seq=6 ttl=64 time=988 ms
64 bytes from 10.255.255.1: icmp_seq=7 ttl=64 time=768 ms
64 bytes from 10.255.255.1: icmp_seq=8 ttl=64 time=1229 ms
64 bytes from 10.255.255.1: icmp_seq=9 ttl=64 time=569 ms
64 bytes from 10.255.255.1: icmp_seq=10 ttl=64 time=705 ms

--- 10.255.255.1 ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 9066ms
rtt min/avg/max/mdev = 569.088/1121.220/2547.054/561.680 ms, pipe 3
root@OpenMPTCProuter:~# 

```
Why so and how to improve it?
[image]

Ysurac, 2022-11-07:
Please use template when you open an issue.
You should update VPS.
Wan address can't be detected correctly on all WAN, there is a configuration problem.

rdmitry0911, 2022-11-08:
I updated VPS and created a new topic using template, with more detailed information

closed 2022-11-08 by Ysurac