Ysurac/openmptcprouter #2645: Notifications when a line goes down

iambenmitchell, 2022-11-08:
**Is your feature request related to a problem? Please describe.**

Sometimes one of my connections will go down and the ISP router will need rebooting. But I don't find out until I notice the connection is slow
**Describe the solution you'd like**

I would like the ability to be notified via email or something similar when a connection goes down. 
**Describe alternatives you've considered**

I've thought about running a speed test on a device even 5 minutes but this is inefficient 
**Additional context**

Ysurac, 2022-11-08:
It's in Services->OMR-Tracker, "Mail alert" checkbox. Mail configuration is in Services->E-mail.

JRTechServices, 2022-12-20:
I am having an issue receiving the mail alert for ISP failure. I am testing by unplugging eth and shutting down vps.    

smtp.mailserver.com
587
Check Box checked TLS
Check Box checked STARTTLS
username redacted@domain.com 
password redacted 
From redacted@domain.com 
To redacted
To 

Mail alert Check box checked. Under Default Settings & Under ORMVPN 

Latest version of OMR, Let me know how I can help.

closed 2023-03-26 by github-actions[bot]