Ysurac/openmptcprouter #2647: Zoom app problem

mvmaksnet, 2022-11-08:
I tried use openmptcp with zoom app, but there are some problems , it looks like connection is changing constantly (I can see zoom stats on other side pc which is also on zoom). Streaming with vmix and obs is ok and it bonds connections very good, but zoom have problem.

I saw this in sys log but dont know what is problem...
mptcp_init4_subsockets: token 0xc8fafd5e bind() to 10.255.254.2 index 24 failed, error -99

  - OpenMPTCProuter version: 0.59.1
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: Maksnet
  - OpenMPTCProuter platform: raspberry pi 3 b+

Ysurac, 2022-11-08:
Please use template when you open an issue.

What is the VPN used ?

mvmaksnet, 2022-11-09:
Glorytun tcp

Ysurac, 2022-11-09:
"10.255.254.2" in log is Glorytun UDP, did you try with it ?
Do you have same problem if you use omr-bypass to bypass zoom connection ?

closed 2023-02-13 by github-actions[bot]