Ysurac/openmptcprouter #2648: glorytun throughput is tooo slow

rdmitry0911, 2022-11-08:
## Expected Behavior
Throughput via glorytun (default vpn for all protocols other than tcp) is about the same as with a regular tcp networking
## Current Behavior
Throughput via glorytun is about 10 times less than a regular tcp networking->

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c 
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1028 
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: RPI4 

I have 3 internet lines: mobile, Starlink and fixed. All 3 are working good. I'm using the default configuration for everything. All the specifics is located only in /etc/config/network and in OMR bypass (both are attached). Nothing special in log files.
However tracepath to 8.8.8.8 from OMR looks like this:
```
root@OpenMPTCProuter:~# tracepath 8.8.8.8
 1?: [LOCALHOST]                      pmtu 1500
 1:  10.255.255.1                                        1050.824ms 
 1:  10.255.255.1                                        2730.835ms 
 2:  192.168.40.1                                        2686.788ms 
 2:  192.168.40.1                                        1754.503ms 
 2:  192.168.40.1                                        778.461ms 
 3:  172.16.44.1                                         1925.886ms 
 4:  static.241.19.21.65.clients.your-server.de          1921.119ms 
 5:  213-239-224-125.clients.your-server.de              2410.352ms 
 6:  core40.sto.hetzner.com                              4663.999ms 
 6:  core40.sto.hetzner.com                              5458.358ms 
 7:  142.250.161.204                                     5774.685ms asymm 11 
 8:  no reply
 9:  no reply
10:  no reply
11:  no reply
12:  no reply
^C
root@OpenMPTCProuter:~#
```
while from VPS it looks like this:
```
root@omr-vps:~# tracepath 8.8.8.8
 1?: [LOCALHOST]                      pmtu 1500
 1:  _gateway                                              0.194ms 
 1:  _gateway                                              0.136ms 
 2:  172.16.44.1                                           0.230ms 
 3:  static.241.19.21.65.clients.your-server.de            0.565ms 
 4:  hos-tr1.juniper1.dc5.fsn1.hetzner.com                 1.049ms 
 5:  core40.sto.hetzner.com                                6.251ms 
 6:  142.250.161.204                                       6.455ms asymm 10 
 7:  no reply
 8:  no reply
 9:  no reply
10:  no reply
11:  no reply
^C
root@omr-vps:~# 
```
Tracepath via individual internet lines also looks fine:

[image]

Strange things on this picture are:

1. The same wan address 192.168.40.1 for all the individual lines,  which is the address of a router **after** VPS
2. Wlan address 172.16.66.1, while it is definitely 10.11.11.1

/etc/config/network:
```
root@OpenMPTCProuter:~# cat /etc/config/network

config interface 'loopback'
	option proto 'static'
[... 411 lines cut ...]

config domains
	option name 'forum.xda-developers.com'
	option interface 'eth681b5c'

config interface 'etha6fd3emac0'
	option id '38'

config dest_port
	option dport 'XXXXX'
	option proto 'udp'
	option interface 'eth681b5c'
	option note 'wg0 via starlink'

config domains
	option name 'google.com'
	option interface 'etha6fd3e'
	option note 'accounts.google.com'

config lan_ip
	option ip '172.16.66.120'
	option interface 'etha6fd3e'
	option note 'lg gram'

```

rdmitry0911, 2022-11-08:
I noticed a strange records in system log:
```
Tue Nov  8 14:36:39 2022 daemon.debug dnsmasq-script[1]: /usr/lib/dnsmasq/dhcp-script.sh: .: line 5: can't open '/usr/share/libubox/jshn.sh': No such file or directory
Tue Nov  8 14:36:39 2022 daemon.warn dnsmasq[1]: script process exited with status 2
```
However, /usr/share/libubox/jshn.sh is there with correct access rights:
```
root@OpenMPTCProuter:~# ls -l  /usr/share/libubox/jshn.sh
-rwxr-xr-x    1 root     root          5457 May 18 01:11 /usr/share/libubox/jshn.sh
root@OpenMPTCProuter:~# 

```
I have no idea if it is related to weak glorytun throughput issue, but it is abnormal too

Ysurac, 2022-11-08:
Your configuration is strange with some macvlan, some direct,... You should start a fresh configuration.
In the tracepath you have "192.168.40.1" multiple times.
The IP shown in status pages for wan is the result of a request to the VPN so this is the IP of the router shown, should not be a big problem.

rdmitry0911, 2022-11-08:
> Your configuration is strange with some macvlan, some direct,...

It is because I have 2 lines on my roof with a direct connection and mobile from iphone I have indoors connected to a lan host. 

> In the tracepath you have "192.168.40.1" multiple times.

It's only with glorytun tcp vpn. With glorytun udp it is totally different:
```
root@OpenMPTCProuter:~# tracepath -n 9.9.9.9
 1?: [LOCALHOST]                      pmtu 1442
 1:  10.255.254.1                                        105.676ms 
 1:  10.255.254.1                                         84.861ms 
 2:  192.168.40.1                                        102.696ms 
 3:  172.16.44.1                                          83.862ms 
 4:  65.21.19.241                                         87.072ms 
 5:  213.239.224.125                                     112.591ms 
 6:  213.239.203.210                                      80.378ms 
 7:  185.1.160.109                                       137.729ms 
 8:  109.200.218.46                                      129.955ms 
 9:  188.122.80.228                                      122.205ms 
10:  9.9.9.9                                             122.031ms !H
     Resume: pmtu 1442 
root@OpenMPTCProuter:~# 

```
It looks good  and pmtu is different.

Maybe mtu with glorytun tcp is calculated wrong way and this wrong mtu is a reason for the problem? 

And a question about /usr/share/libubox/jshn.sh May it be tied to this problem, or it is a different bug and I have to open another issue?

Ysurac, 2022-11-08:
You can try to set a lowest MTU on tun0 and check: ```ip link set mtu 1440 dev tun0``` via SSH on the router.
For the jshn error, try a ```chmod u+x  /usr/share/libubox/jshn.sh```

rdmitry0911, 2022-11-08:
> You can try to set a lowest MTU on tun0 and check: `ip link set mtu 1440 dev tun0`

I did a try. No help. 

> You should start a fresh configuration.

Is there a way to start over without a factory reset as I have some packages installed and I don't want to reinstall them all?

Ysurac, 2022-11-09:
No but you can save installed package list when you do a backup.

rdmitry0911, 2022-11-09:
> No but you can save installed package list when you do a backup.

That's strange. Everything is written in config files. The packages itself are the regular ones. Why don't remove the config files related to the omr configuration leaving all the rest files in places? I'm going to reinitialize omr remotely without having direct access to the system and make a factory reset might leave me without an access to the system. Clearing config files would be much more safe procedure in this case.

closed 2023-02-13 by github-actions[bot]